arkypita/LaserGRBL #640: Image import - EXIF gives a wrong size after roting the image 90°

tomturican, 2019-05-06:
When i import an image and rotate it 90°, the EXIF reading give me a wrong size. (tested with 2400 dpi jpg and png file). I use the 3.0.10 Version.
While importing my PNG Image, the dpi value from EXIF reading is one point to low. In my case 2399 dpi instead of 2400 dpi.
I will use LaserGRBL to exposing my PCB (with a 50mW UV Laser), exported from KiCad. First tests looks very good, but i need exact dimensions.....

arkypita, 2019-05-07:
For what I can see, it work

https://youtu.be/oGe95z92wJg

arkypita, 2019-05-07:
If you rotate the image using LaserGRBL rotation button it work. Tested now with latest version (no code changes in this part).

If you use windows or an external program to rotate the image, he is probably the cause, maybe this deletes the exif data from file? Please check file property.

arkypita, 2019-05-07:
Also tested with PNG. No difference between rotated/unrotated.
Only rounding error but it is an issue of png file format.

Look at this: https://forums.adobe.com/thread/2139680

closed 2019-05-07 by arkypita

tomturican, 2019-05-07:
Hello & thanks for the answer,

pls look on my screeenshots. The rotating was done within lasergrbl. The autosize Funktion should only swap the  values (x/y). This file is also attached.

Greetings
tom

Von: arkypita
Gesendet: Dienstag, 7. Mai 2019 10:33
An: arkypita/LaserGRBL
Cc: tomturican; Author
Betreff: Re: [arkypita/LaserGRBL] Image import - EXIF gives a wrong size after rotating the image 90° (#640)

Also tested with PNG. No difference between rotated/unrotated.
Only rounding error but it is an issue of png file format.
Look at this: https://forums.adobe.com/thread/2139680

arkypita, 2019-05-07:
Can't see any screen shot

tomturican, 2019-05-07:
I have attached them on this mail….

Von: arkypita
Gesendet: Dienstag, 7. Mai 2019 18:17
An: arkypita/LaserGRBL
Cc: tomturican; Author
Betreff: Re: [arkypita/LaserGRBL] Image import - EXIF gives a wrong size after rotating the image 90° (#640)

Can't see any screen shot

tomturican, 2019-05-07:
[image]
[image]
[image]

tomturican, 2019-05-07:
the dpi value itself is ok. but the B & H values (width and hight) dont swap after rotating.

arkypita, 2019-05-07:
Understand. Try check/uncheck autosize please

arkypita, 2019-05-07:
I will test as soon as possible

tomturican, 2019-05-07:
check/uncheck tested, but no change....

Thanks ;-)

arkypita, 2019-05-08:
fixed

tomturican, 2019-05-08:
Great, it works perfect!

Many thanks!

arkypita, 2019-05-09:
You are welcome, thanks for bug report.

commit 3733667d (2019-05-08): fix #640

Fix issue with image size calculation from exif when image was rotated or cropped.