arkypita/LaserGRBL #642: Power management of the laser is incorrect

zahidhasanmim, 2019-05-09:
Hello arkypita.
Thanks a lot for such a powerful software.
I am new here.

I am facing problem with laser power management option in bottom right corner.My  laser gives 50% power when I select 1x power option and gives 100% power when i select 2x power option.The software by default starts with 1x power.Every time I open this software I have to select 2x power option.It is too much annoying because I sometime forget to select 2x power and my laser starts working at 50% power by default.
[image]
[image]

It would be too much helpful for everyone if the laser power management option is made 100% for 1x and it is set as default laser power every time I reopen this software.

arkypita, 2019-05-09:
You are wrong. What you call "power management" is a power override, that could be used to real-time change power (and speed) during engraving process. 

They are a new feature of Grbl v1.1 called "[real-time override](https://github.com/gnea/grbl#update-summary-for-v11)" ([more info](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#extended-ascii-realtime-command-descriptions)) and LaserGRBL offer an interface on it https://lasergrbl.com/it/usage/overrides/

By default there are no override so power command like S255 and speed like F2000 are executed as required by GRBL firmware. 

If you are experiencing laser power level lower than expected please check your GRBL configuration for $30

Read this: https://github.com/arkypita/LaserGRBL/issues/620#issuecomment-487269442

zahidhasanmim, 2019-05-09:
I am attaching my configuration.Can you find my mistake?
[image]

[image]

arkypita, 2019-05-09:
I cannot find any mistake, all seem to be ok.

zahidhasanmim, 2019-05-09:
But i am getting half power for 1x and full power for 2x

arkypita, 2019-05-09:
You have very low acceleration. I suggest to use same values of maximum rates.

arkypita, 2019-05-09:
If you use M4 could be an effect of Dynamic Laser power mode.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

zahidhasanmim, 2019-05-09:
ok. I think i need to do more expriment

arkypita, 2019-05-09:
Whould you like to try engraving this sample? [test1.zip](https://github.com/arkypita/LaserGRBL/files/3162845/test1.zip)

This should produce 3 lines.
[image]

First line is engraved using $32 laser mode OFF and M3 constant power
Second line using $32 laser mode ON and M3 constant power
Third line using $32 laser mode ON and M4 dynamic power

Please take note of F values and S values shown by LaserGRBL in upper-right angle of its windows.
They should be F1300 and S255.

If they F value is lower it means that your $$ configuration is slowing down the real speed because of some configured limit (speed or acceleration). If S is lower it means that M4 dynamic power is scaling down the power because of that limit.

If you want post picture of engraving result.

closed 2019-11-03 by arkypita