arkypita/LaserGRBL #643: continuous Jogging move only to one direction

oliveritech, 2019-05-09:
Hello, thank you for this new 3.0.13 release. If I enable continuous Jogging my machine is moving only towards the + space; if continuous Jogging is off everything works as is should. Is it only me with this issue?

arkypita, 2019-05-09:
Hi @oliveritech continuous jogging was developed as a new feature, so it could have some youth problem :-)

Please exactly describe the problem, so I can work on it.

StuartB4, 2019-05-09:
Working fine for me, all directions and Home, also stops as it should when it reaches the maximum number set in GRBL $130, $131, $132 settings.

oliveritech, 2019-05-09:
Hi, basically if continuous Jogging is enable my machine move only forward and can't move back with the arrow keys.

arkypita, 2019-05-09:
I use limit switches + homing, so when I power on I do homing and homing procedure will seek for switches and set zero point in the lower-left limit of my machine. Is not possible for me to go negative.

I think that @oliveritech does not use homing, so it have zero point in the position where the laser is when he turn on, so he need to use negative space.

oliveritech, 2019-05-09:
I have my machine set exactly the same with the Homing in to the lower left and limit switches on X and Y Axes, no Z axes enabled. I flashed the GRBL 1.1f in to the Arduino and is working now !
Is it normal that with continuous Jogging enable the Home button in the middle of the arrows, now need to be kept clicked to go home and not just one click like before ?

arkypita, 2019-05-09:
> I have my machine set exactly the same with the Homing in to the lower left and limit switches on X and Y Axes, no Z axes enabled. I flashed the GRBL 1.1f in to the Arduino and is working now !

Which version you had before?

> Is it normal that with continuous Jogging enable the Home button in the middle of the arrows, now need to be kept clicked to go home and not just one click like before ?

Yes, for homogeneity.

oliveritech, 2019-05-09:
0.9
Ok thank you for your great work.

closed 2019-05-09 by oliveritech

mschaeck, 2019-06-01:
I use GRBL 1.1f and continuous Jogging is not woking (only in one direction)
G53Y390.0F1960 for one direktion works fine, in the other direction the command the software dens G53Y0.0F1960 is not working (is moving in the same direction).
Why is not G53Y-390.0F1960 sendet?

oliveritech, 2019-06-01:
Hello mschaeck, LaserGrbl mouve only in the positive direction. Basically the way I understud and resolved is; after you sucessfully home your machine with limit swithces, this shold be X0Y0 and from there you could mouve your machine only in the + space and not in the - unlike pro machine that from homing mouve to the - space. This should help https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to
ciao

mschaeck, 2019-06-01:
I understand, I am using the home function. I home the machine with limit switches. Why is in the right upper conrer still not X0Y0 after homing?

oliveritech, 2019-06-01:
Because that is where in the GRBL Firmware the $23 set the homing cycle direction , but that is not the origin of you X0Y0

oliveritech, 2019-06-01:
https://openbuilds.com/threads/confused-about-homing-machine-position-zeroing-etc.6848/

mschaeck, 2019-06-03:
[image]
I guess the first problem is why after homing the machine with limit switches (wich works fine) the first line with the absolut position is not set to zero, see the image.

mschaeck, 2019-06-03:
I have set $23 = 1