arkypita/LaserGRBL #644: laser don't turn on/off

alexdam92, 2019-05-11:
Hi i'm Alex from Italy and i have a big issue with my laser... I hope that someone can help me because i read all discussion on this site but i didn't found any answer. 
I can't turn on/off my laser.    My laser is https://www.banggood.com/EleksMaker-FB03-500-500mW-Blue-Laser-Module-2_54-3P-TTLPWM-Modulation-for-DIY-Laser-Engraver-p-1266178.html?rmmds=myorder&cur_warehouse=CN . I have a cnc shield v 3.0 and i installed firmware grbl V 0.9I. The motor move very well and i don't have any kind of issue, but when i send the command M03 with laser GRBL my laser don't turn on! I have tried to connect PWM cable to z+ , z- and SpnEn without any result! The only way to turn on my laser is to press on the little button on laser driver, but it is always on/off! Anyone know what is the ptoblem? it's possible that my cnc shield is not compatible with grbl 0.9? 
[image]
[image]
[image]
[image]
[image]

alexdam92, 2019-05-11:
I have the lastest version of laser grbl, and i also tried with grbl 0.8 and 1.1 version.

StuartB4, 2019-05-11:
Try connecting a ground lead from the CNC shield  (one of the pins in the black strip next to the Z+ pin) to the laser driver board ground.
There single Red lead you have coming from the Laser driver board PWM / TTL should go to the Z+ or Z- (they are both linked together so it does'nt matter which you choose)  But you have a Green lead on the Z+, where does that go ?

alexdam92, 2019-05-12:
thanks stuart B4!! It worked fine for me :) can i ask you if i need some mod to the firmware version 1.1  to work with my laser? or can i just download it and flesh on arduino? Thanks Alex

StuartB4, 2019-05-12:
Just download it and flash it. It should work as it is.
You can configure it how you want after.

alexdam92, 2019-05-13:
thanks so much! :) just the last question.. do you think that  microstep are useful with this tiny machine? or 
you think it could make the print worse for some problem? now i'm using 5 step/mm and it is working...

StuartB4, 2019-05-13:
If it's working ok and you are happy with the results then leave it at that.
You could experiment with different microsteps and burn the same image at all 
the different settings just to check the best quality.
But then there are also the LaserGRBL settings which can affect quality, like lines per mm, laser power, egraving speed, brightness / contrast etc.

alexdam92, 2019-05-15:
Thanks again so much Stuart! have a nice day ;)

closed 2019-05-15 by alexdam92