arkypita/LaserGRBL #645: Perfect circles with EleksMaker A3 Pro?

maillemaker, 2019-05-14:
Is anyone with an Eleksmaker A3 pro able to burn a perfect vector circle?

Steve

RonyZlin, 2019-05-14:
Yes, I do. I think the most important thing is the correct tension of the belts.

maillemaker, 2019-05-15:
So how tight should they be?  I have tried so tight it stalls out the motors and everything in between.

Can I get a copy of your settings for your a3 pro to re-load into mine?

Steve

maillemaker, 2019-05-15:
And, what are your laserGRBL settings?

RonyZlin, 2019-05-16:
Settings:

[GRBL_Configuration_Rony_2.txt](https://github.com/arkypita/LaserGRBL/files/3188994/GRBL_Configuration_Rony_2.txt)

[image]
[image]
[image]

RonyZlin, 2019-05-16:
Some videos:
[Belt tension](https://youtu.be/Dq7W-a9pWHg)
[video1](https://youtu.be/og6RIotnSc8)
[video2](https://youtu.be/dJ486orKkOc)

RonyZlin, 2019-05-16:
I found it's not perfect, but the shift is so small that I don't mind (approx. 
0.1 mm). I didn't notice it at all in usual engraving.

[image]

brakthehun, 2019-05-17:
How do you generate a circle?? Is it a picture?? Is it a G2 / G3 command??

RonyZlin, 2019-05-17:
It was a black bitmap filled circle. In LaserGRBL: Conversion tool: Vectorize, Filling: none, Laser ON: M4

StuartB4, 2019-05-17:
RonyZlin.

You can view and save the code of any image you have prepared by going to the File menu and select Save Program.

brakthehun, 2019-05-17:
Try using this instead in the command line:
G92X0Y0
G0X10Y60
F500
M4S255
G2X10Y60 I50J0
M5

This will create a 100mm circle using the arc command (G2)

maillemaker, 2019-05-18:
I have discovered my problem.  The problem was 2 broken leads on the bottom side of the left gantry motor.  Once I repaired the broken leads, the machine now works fine.

broken leads:
[image]

Test after fixing:
[image]

closed 2019-05-18 by maillemaker