arkypita/LaserGRBL #650: one axis not working as expected

m3vuv, 2019-05-16:
.Hi all,i need some help with a homebrew cnc laser engraver,to start its using an arduino with a cnc sheild running gbrl 1.1f,1.25mm pitch leadscrews driven by nima 24 steppers,im trying to use lasergbrl,it jogs fine on both axis,but if i try to engrave an image after lasergbrl has converted it, one axis hardly moves,all i get is a thick black line burnt in the workpiece,it does the same whether its a grayscale or vector etc,same with the engraving direction,if i say swap the direction the problem swaps axis also,if that makes sense,also it goes in fits and starts and loses sync with the curser on the screen,its like its waiting for a buffer to fill up,ive tried all different conection settings ie syncronous buffered etc,it makes no difference,ive decoupling caps everywhere  and have tried a few different usb cables,if i use ugs and run a gcode with that it works fine,im not very impressed with lasergbrl to say the least so far,just seems good for producing scrap with as many freezes glitches and bugs,san anyone advise before the lot goes in the skip??,regards m3vuv,ps im using a uno and laser mode is enabled.this is driving me nuts!!!!

m3vuv, 2019-05-16:
well i tried this test file from here,that works fine,im having issues with a png to vector conversion,i will attach both,can someone change the png to a vector for me,if they can and it works then i know where to look,cheers m3vuv.
[image]
[test.nc.txt](https://github.com/arkypita/LaserGRBL/files/3187831/test.nc.txt)

m3vuv, 2019-05-16:
please help im at my wits end!

StuartB4, 2019-05-16:
Just load you png and select Vectorize.

[image]

m3vuv, 2019-05-16:
ive done that but still seems to screw up,ive exahausted just about everyway of tring to turn it to working gcode,otherwise i wouldnt ask someone else to convert test and upload it.

m3vuv, 2019-05-16:
also meant to say everytime it pauses for whatever reason it does it with the laser on,burns big chunks of the work piece

m3vuv, 2019-05-16:
it stops about every 20 seconds for about 5 or 6 seconds when engraving a job,

StuartB4, 2019-05-16:
> ive done that but still seems to screw up,ive exahausted just about everyway of tring to turn it to working gcode,otherwise i wouldnt ask someone else to convert test and upload it.

What are you trying to say. What are you ending up with?
If you load an image and click Vectorize it will give you the image to engrave in the correct format.
like in the image I posted above.
Then you can save the GCode using the File menu, Save Program. 
If that does'nt work for you then you must have some setting wrong somewhere.
Eithe LaserGRBL, GRBL or even your machine setup.

Can you post pictures of the settings you are using, and your GRBL configuration settings.
Have you tried other laser engraving software besides LaserGRBL?
You could try T2Laser. They have a free trial version. So you can try it.
http://t2graphics.weebly.com

m3vuv, 2019-05-16:
im ending up with firewood so farheres the settings etc and some pics

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[20190516_225514.zip](https://github.com/arkypita/LaserGRBL/files/3189227/20190516_225514.zip)
[Pictures.zip](https://github.com/arkypita/LaserGRBL/files/3189231/Pictures.zip)
[Uploading Pictures.zip…]()

[Uploading 20190516_225514.zip…]()

m3vuv, 2019-05-16:
i think thats all,im just waiting for the next piece of firewood to finish lol,when i try and jog 50mm it will run for 11 seconds then stop for 7 seconds then continue,it exhibits the same behavior engraving as well but with the laser powered full time when it stops,seems like its waiting for buffers to refil?,heres the latest abortion done with the above settings.
[20190516_233608.zip](https://github.com/arkypita/LaserGRBL/files/3189251/20190516_233608.zip)

brakthehun, 2019-05-17:
What version of windows are you using??

m3vuv, 2019-05-17:
win7 ultimate

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

brakthehun, 2019-05-17:
Use the command line and give it custom g code commands.
First,, set zero (Globe with a pin button)
Try:
G0X20Y20
and then:
G1X0Y0 F500
Does it move diagonally?? does it move about 20mm up and 20mm to the right??

m3vuv, 2019-05-17:
i will try it later its 4am here,do you mean the comand line in lasergbrl
or the c prompt in windows?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

brakthehun, 2019-05-17:
command line in laserGRBL

m3vuv, 2019-05-17:
when i issue the first line of code both axis move but carry on and dont
stop untill i pull the usb cable and lasergbrl has frozen by that time.

m3vuv, 2019-05-17:
if i reset the arduino and reconect,then issue the 2nd line of code,i assume both axis should retern to zero?,they both continue in the same direction as the first command but againe carry on and dont stop untill i unplug the usb cable,it does the same with ugs as well as laser gbrl.

m3vuv, 2019-05-17:
ive tried 2 usb cables and have 100nf and 47uf caps on the power supply lines to the cnc shield,the wires to the steppers are not shielded,could this be the issue?

StuartB4, 2019-05-17:
Which cnc shield do you have? 
If it's a protoneer, do you have heatsinks on the stepper motor driver chips, they may be overheating if not.
Also, if it is a protoneer, did you set the current limits on the driver boards for the motors?

m3vuv, 2019-05-17:
hi the cnc shield is a protonier ver3 with heatsinks on the drivers with
current set at 500ma and there fan cooled.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

StuartB4, 2019-05-17:
Enable PWM Mode in LaserGRBL settings. Then try changing M3 to M4 in Target Image settings.
Also in GRBL Configuration settings change $120 and 121 to 10, 2 seems a bit low.

[image]

[image]

[image]

m3vuv, 2019-05-17:
ive already tried that,same results.

StuartB4, 2019-05-17:
Sounds more like you have a hardware problem than a LaserGRBL problem.

StuartB4, 2019-05-17:
Can you try with a different Arduino UNO.

m3vuv, 2019-05-17:
done that and tried changing the cnc shield,its one reason why i asked for the png image to be converted and reuploaded,would of helped my trouble shoot but no one bothered!

StuartB4, 2019-05-17:
Why not use a free online png to vector converter. Just look on google.

StuartB4, 2019-05-17:
This one is good:
https://image.online-convert.com/convert-to-svg

m3vuv, 2019-05-17:
that may work but isnt the solution as i see it,something is obiously wrong with my setup,if i use an online converter to me thats just another unknown into the equation!

m3vuv, 2019-05-17:
well as i last resort im swapping all the stepper cables with multicore shielded flex gronded on the shield at each end,should have it back up and running in the morning.

brakthehun, 2019-05-17:
It sounds like your internal settings. Type in $$ and take a better screenshot.

m3vuv, 2019-05-18:
my max feedrate is 500mm/min,seems like a dangerous bug!

brakthehun, 2019-05-18:
What is your x and y resolution??

m3vuv, 2019-05-18:
do you mean steps per mm?,if so there both 2560,im using 1/32 microstepping with 1.25mm leadscrews and 1.8deg steppers.

m3vuv, 2019-05-18:
well an update,all the cables to the steppers are now shielded and grounded,it seems better but the big issue is each axis tops halfway thru running a command but with the laser on,its leading to a burnt mess!,this has been an issue doing a long jogg as well,ive mentioned this previously but ppl seem to ignoor it!

sinfocomp, 2019-05-18:
the problem to me looks like it is becouse using a leadscrew requires lots of pulses at a speed the arduino is not capable of, try to set your stepper driver at its minimal microsteps, also make sure youre not using the USB ports for other stuff since you can under-run your buffer, i would use a belt drive for laser since the leadscrew force is not needed
another thing is wour laser C02 or diode ?
if it is C02 make sure you have optoisolation on the control lines on the arduino side to prevent interference, also locate your laser psu at least 3 feet away from the arduino, 
i have used this software and arduino on large 300watt co2 laser machines with no problems at all
my only no no is for plasma, never got rid of the HF interference
hope this helps

sinfocomp, 2019-05-18:
ohh one more thing if youre using pololu drivers set the current at its minuimum otherwise the drivers will overheat and cease operation for some seconds then continue probably causing the scorching you described

m3vuv, 2019-05-18:
its a nichia m-140 diode,laser,the drivers seem cool,i already looked at that,the arduino is the only thing plugged into the pc.

m3vuv, 2019-05-19:
the buffers getting empty sounds like the cause,i have mentioned that in previous posts,no one seemed to bother with it tho,i think youve hit the nail on the head,i will try 16/ steps,in theory should need half as many pulses i imagine?.

m3vuv, 2019-05-19:
well ive gone to1/16th steps,better but noisy as hell issue now is anything bellow 70mm min speed wakes the dead 7o and more throws a fault up with the board and streaming stops,its one thing after another also if i do a continous jog for more than about 2 seconds it wont stop traversing unless i unplug the usb and power,then when i reconect it wont find the ports have to exit laser gbrl and reset the uno,so have to go thru it all againe,at the moment laser gbrl doesent seem fit for use!!,sory but my experiance so far,think i got more chance of winning the lottery than this software ever working without issues tbh.

m3vuv, 2019-05-19:
poxy thing is getting nearer to the dumpster!!/skip.

StuartB4, 2019-05-20:
Have you tried updating .Net Framework for Windows?
Have you tried uninstalling LaserGRBL and reinstalling it?
Have you tried reflashing GRBL to the Arduino?
Have you tried changing the Stepper drivers on the shield?
Which stepper drivers are you using?
If you have the A4988 red board ones change them to DRV8825.
They can handle a bit more current.
How powerful is your power supply, can it provide enough amps?

m3vuv, 2019-05-20:
im using drv 8835 drivers with a psu at 28v 12amp,yes its the latest .net if thats what you mean,not reflashed grble yet or reinstalled lasergbrl yet tho,will do that later as its 4am here.

brakthehun, 2019-05-20:
max pulserate for Arduino GRBL is 30khz (30,000 pulses per second).
2560 steps per mm x 500mm/minute = 1,280,000 pulses per minute.
1,280,000 pulses per minute / 60 seconds = 42,666.666.
That pulse frequency will lock the arduino in a loop.

m3vuv, 2019-05-20:
well suprise suprise,reinstalled ,net framework laser gbrl,just did the first job that worked!

m3vuv, 2019-05-20:
heres a pic
[image]

m3vuv, 2019-05-20:
only issue now is if i try to do any fill gbrl falls over a few mins into the job.any ideas

arkypita, 2019-05-20:
> only issue now is if i try to do any fill gbrl falls over a few mins into the job.any ideas

1) Read previous @brakthehun reply. More info [here](https://github.com/grbl/grbl/issues/41)
2) Electric noise or chinese arduino clone, please consider to buy an original one. More info [here](http://lasergrbl.com/it/faq/)

Workaround for n°2 is set a threading mode Quiet for LaserGRBL as in faq

m3vuv, 2019-05-27:
well,it works using half steps,the issue is it makes enough noise to wake the  dead,is there any other arduino model with a faster clock speed that will work with a protoneer cnc shield  that i can use 1/32 microstepping with?,cheers m3 vuv.

brakthehun, 2019-06-06:
Can you take a picture of your machine with a close up of the leadscrews??
I think you can speed it up with a different set of parts.
What steppers are you using??

closed 2019-11-03 by arkypita