arkypita/LaserGRBL #651: Problem with image size: jpg, gcode & inkscape

AdamMulica, 2019-05-16:
Hello, i have problem with my laser co2. I don't understand why program LaserGRBL change image size when i use gcode or svg. What i must change, what parameters in program LaserGRBL or in Arduino Atmega328p with grbl_v1.1f or Inkscape?

Thanks very much for all your time and help.

[image]

[image]

[image]

brakthehun, 2019-05-17:
You can choose your image size in the second window of image process. They're always in mm.

AdamMulica, 2019-05-17:
I finded the solution :D
I musted changes parametrs 
$100 and $101 - standard 250,000 - now 10mm is 1cm
$110 and $111 - for faster travel of the x and y axes - standard 500,000
$120 and #121 - standard 10,000

huh :D

[image]

closed 2019-11-03 by arkypita