arkypita/LaserGRBL #652: Rx, Tx, and FTDI232

Cipriano23, 2019-05-16:
i couldnt find any info related to this so, before buying anything, wanted ask if someone tried it.
its possible to use Rx and Tx pins through a ft232 converter instead of usb jack?

m3vuv, 2019-05-16:
ive done that to program pmr radios from rs232 to ttl i think it was,if i remember ttl is 5v logic levels and 232 is 9v,i used a max 232cp ic

Cipriano23, 2019-05-16:
yea, the converter isnt the problem, i just wanna know if i use Rx and Tx, LaserGRBL will detect it and work like it does with the usb connection on arduino.

arkypita, 2019-05-18:
I think is possible.
When I use esp8266 to connect grbl via telnet i conenct arduino rx-rx to esp bypassing usb, so it should be ok to do the same with an ft232 as soon LaserGRBL can see it as a COM port.

Cipriano23, 2019-05-19:
okay then, i already bought the converter. as soon as it arrives ill try it and check if it works.

David943, 2019-05-20:
well, i bought the yp-02 converter instead of the rs232, i checked both datasheet's and they are similar.
after making the connections, lasergrbl doesnt recognise the Arduino.
i see the converter as a COM port but when i connect it, i dont get the msg that grbl is detected.
my connections, 1- rx to tx, 2- tx to rx, 3- 5v, 4- GND
ive used a jumper on converter, 5v with VCC.

[image]

any ideas?

Cipriano23, 2019-05-20:
well, i bought the yp-02 converter instead of the rs232, i checked both datasheet's and they are similar.
after making the connections, lasergrbl doesnt recognise the Arduino.
i see the converter as a COM port but when i connect it, i dont get the msg that grbl is detected.
my connections, 1- rx to tx, 2- tx to rx, 3- 5v, 4- GND
ive used a jumper on converter, 5v with VCC.

[image]
(this is the msg that doesnt appear)

any ideas?

closed 2019-11-03 by arkypita