arkypita/LaserGRBL #653: How can I connect laser module to cnc sheild v3.0

Timmerhus08, 2019-05-17:
Hello every one,

 I am new and I need help to connect   **Laser-Modul 500mW 405nm blau-violett 5V 0.5W** 
 ( 2 wires Read and Black )to cnc sheild v3.0 with

 arduino uno. I have connected the Read wire from the laser to Z+ in cnc shield but nothing happend and then I used this circute in the picture and nothing happend also , if any one can take me to the right direction?.

[image]

Best Regards

brakthehun, 2019-05-17:
Red wire needs to be connected to 5v pin. 
560 ohm line needs to be connected to Z+

Timmerhus08, 2019-05-17:
Tanks for your answer,

I did it but nothing happend and when I connect the laser direct to the 5V Power supplay it is working.

brakthehun, 2019-05-18:
You need 3 connections,, right?? Black to GND,, Red(Rot) to 5V,, and the 560 ohm line to Z+??

brakthehun, 2019-05-18:
Can you take a pic of your setup??

Timmerhus08, 2019-05-18:
Tank you so much, I made like what you said and worked but when I try to engrave a pic the lser seen very weak and when I connect the laser to external powersupplay 5v it is strong when I moved manuel and brun the cardboard.

Timmerhus08, 2019-05-18:
I am tryied to Upload a pictures but i couldn't .

Realy Thanks.

Timmerhus08, 2019-05-18:
[image]
[image]
[image]

Timmerhus08, 2019-05-18:
[image]
[image]

brakthehun, 2019-05-25:
It looks like your transistor is miswired:
My transistors - go like this from the front view (heat sink plate on the back):
signal (z+ / resistors)
drain (laser (-))
ground (ground)

Timmerhus08, 2019-05-25:
thank you so much, you have right (miswired) I did same your opinion and worked perfectly but it is not burn.  When I connected the laser direct to the power supplay its burning but connected throw the circuit its weak.
Best Regards

brakthehun, 2019-06-06:
I think this is solved.

Timmerhus08, 2019-06-06:
Yes thanks but it is weak not burn.

brakthehun, 2019-06-06:
What laser module is it?? It looks like a 100mw red,, which might not have enough power to burn.

brakthehun, 2019-06-06:
I'm looking at the pics. It looks like it needs to be focused better.

StuartB4, 2019-06-06:
He said it's a " Laser-Modul 500mW 405nm blau-violett 5V 0.5W"

Timmerhus08, 2019-06-06:
please can you give me a link to buy it?.

Best regards

Timmerhus08, 2019-06-06:
Thanks to you brakthehun my laser (Laser-Modul 500mW 405nm blau-violett 5V 0.5W").I tryied to focuse it but same result.
Best Regards

brakthehun, 2019-06-06:
https://www.ebay.com/itm/405nm-500mw-12V-Blue-Laser-Module-for-DIY-Laser-Engraving-Machine-with-TTL/192396147963?hash=item2ccbb43cfb:g:~VQAAOSw0JFbZ~wF

I never know the quality or lifetime of these.

Timmerhus08, 2019-06-06:
Thank you so much.

closed 2019-11-03 by arkypita