arkypita/LaserGRBL #654: Recommended speed for Vectorized and Filled art on pine wood?

maillemaker, 2019-05-18:
Hi all,
I am doing some experimenting with different burn speeds now.  I first tried 500 border and 1000 fill with 8 lines per mm.  This worked pretty good but resulted in some "bleed" of scorching into the surrounding wood.
I now bumped up to 1000 border and 1500 fill and 6 lines per mm and am having another go.  This looks like it might be a little light.

Does anyone have any suggestions for laser engraving pine wood with vector and fill artwork?
Thanks,
Steve

StuartB4, 2019-06-14:
Have a look on at adding Air Assist to your machine.
This is with and without it.

[image]

maillemaker, 2019-06-18:
That is amazing!  I had already taken the step of pointing a small fan at the work area.  This helped, but also pushed the remaining "shadow" all in one direction! :)

I'll go google air assist now but do you have any links?

Steve

arkypita, 2019-06-18:
Also covering with masking tape helps giving clear result

closed 2019-11-03 by arkypita