arkypita/LaserGRBL #655: $32 cant find in grbl setings

geduxk, 2019-05-18:
hi 
I'm trying to run laser but cant find $32 to on laser all time first i was installed 0.9 version i read it does not support that, so find and install 1.1f but still not in list, in configuration.h i end comment two lines 
#define USE_SPINDLE_DIR_AS_ENABLE_PIN
#define SPINDLE_PWM_MIN_VALUE 5 but still doesn't shows if i connect to lasergrbl it shows 0.9f version it doesn't seams logic how it should be, then trying to delete all eprom and flash again but still 0.9j i confused how it flash 0.9j if i downloaded 1.1f

StuartB4, 2019-05-18:
What you can do is just download the GRBL 1.1 .hex file and use a free little program called XLoader to flash it.
It should then work and you can change any setting you need to in LaserGRBL using GRBL Configuration on the GRBL menu.

geduxk, 2019-05-18:
> 
> 
> What you can do is just download the GRBL 1.1 .hex file and use a free little program called XLoader to flash it.
> It should then work and you can change any setting you need to in LaserGRBL using GRBL Configuration on the GRBL menu.
> 
> XLoader is here:
> http://russemotto.com/xloader
> 
> GRBL 1.1f .hex here:
> https://github.com/gnea/grbl/releases

It Works !!!!!!!! thank you

closed 2019-05-18 by geduxk

StuartB4, 2019-05-18:
Thats good to know. :+1: 
Thanks.