arkypita/LaserGRBL #656: Lasergrbl Not Connecting

chucky0735, 2019-05-18:
I click the connect button 5 - 10 seconds later 

benbox laser 3.1(c) mul 2015.06.06
start

error log

18.05.2019 18:23:15	SetStatus   	Machine status [Connecting]
18.05.2019 18:23:15	OpenCom     	Open COM5 @ 115200 baud 
18.05.2019 18:23:25	OpenCom     	Connection timeout!
18.05.2019 18:23:25	CloseCom    	Close COM5 [CORE]
18.05.2019 18:23:25	SetStatus   	Machine status [Disconnected]

arkypita, 2019-05-18:
http://lasergrbl.com/it/faq/

chucky0735, 2019-05-18:
I wonder if it works with my motherboard ? 

my laser machine

https://www.banggood.com/5065cm-Engraving-Area-3000MW-Laser-Engraving-Machine-DIY-Kit-Desktop-Laser-Cutting-p-1268114.html?rmmds=myorder&cur_warehouse=CN

StuartB4, 2019-05-18:
Did you install the CH340 driver for the USB connection?

StuartB4, 2019-05-18:
Just wondered, what is the quality like of the actual machine, is it good quality for the price?
I was looking at that machine a couple of days ago.

chucky0735, 2019-05-19:
yes i got it. But when you open the program (lasergbrl) benbox laser 3.1 (c) mul 2015.06.06. I think it is not compatible with lasergrbl.

By the way machine is not bad. 15w laser think.

StuartB4, 2019-05-19:
Have you had a look in Windows Device manager to make sure your machine shows up there?
Connect machine to usb and switch it on, then check in Device manager.

[image]

arkypita, 2019-05-19:
> yes i got it. But when you open the program (lasergbrl) benbox laser 3.1 (c) mul 2015.06.06. I think it is not compatible with lasergrbl.

I confirm that if your engraver control board comes with benbox then it is not usable with LaserGRBL "as is". If you want to use it with LaserGRBL you should convert it to GRBL by flashing grbl firmware on it. You have to do a little hack on it, but nothing impossible. Google for that, i think you will find some guides.

chucky0735, 2019-05-19:
[image]

My Drives. Benbox Not Problem working but i want lasergrbl :(

chucky0735, 2019-05-19:
> > Evet anladım. Ancak programı açtığınızda (lasergbrl) benbox laser 3.1 (c) mul 2015.06.06. Lasergrbl ile uyumlu olmadığını düşünüyorum.
> 
> Oymacı kontrol kartınız benbox ile birlikte gelirse, LaserGRBL "olduğu gibi" kullanılamaz olduğunu onaylarım. LaserGRBL ile kullanmak istiyorsanız, üzerinde grbl firmware yanıp sönerek GRBL'ye dönüştürmelisiniz. Biraz kesmek zorundasın ama imkansız değil. Bunun için Google, bazı rehberleri bulacağınızı düşünüyorum.

Can you help me ? 
[image]

StuartB4, 2019-05-19:
One option is to buy an Arduino Nano and flash it with GRBL.
Then take out the one on your control board and place the new one in.
That way you can keep the original Benbox firmware on the old board and use the new one with GRBL and LaserGRBL.
If you need Benbox again just swap them back.

chucky0735, 2019-05-20:
> Bir seçenek bir Arduino Nano satın almak ve GRBL ile yakmak. 
> Ardından kontrol panelinden 
> bir tane çıkartın ve yenisini yerleştirin. Orijinal Benbox ürün yazılımını eski kart üzerinde tutabilir ve yenisini GRBL ve LaserGRBL ile kullanabilirsiniz. 
> Eğer Benbox'a tekrar ihtiyacınız varsa, sadece onları değiştirin.

My friend, I live in Turkey. I found this product clone this product. Will this product work?

StuartB4, 2019-05-20:
It should work, but I have never tried with a clone, I've only used genuine Arduino boards.

chucky0735, 2019-05-20:
> İşe yaramalı, ama bir klonla hiç denemedim, sadece orijinal Arduino panoları kullandım.

Okey ı understand. Thank you Bro

closed 2019-05-28 by arkypita