arkypita/LaserGRBL #658: Auto cooling

arkypita, 2019-05-19:
Can someone test this feature?
https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.15

@m3vuv
@brakthehun

m3vuv, 2019-05-19:
i would love to try this but i dont fancy producoucing more firewood,its buggy enough as is!!

StuartB4, 2019-05-20:
Don't have a cooling problem myself because I only have a 2 Watt laser which has been on continuously for 3 to 4 hours at a time, depending on the image.

Just thought  I'd give it a try.
Tried 3 different images with different On - Off time settings.
Seems to work perfectly.
Another good addition to an already brilliant program.

brakthehun, 2019-05-25:
I will try it tonight!!

brakthehun, 2019-05-26:
I did a slow greyscale engraving. Turned out well. I'm a little suspicious of some lines,, but It worked well.
I did a vector created from Inkscape and it took 3 minutes and it paused correctly.

I set it to run for 1 minute,, pause for 1 minute and I set up a 55 second run and told it to run 10 times. It never stopped to cool.

I had one problem - I did a raster and it never paused: 
[image]

brakthehun, 2019-05-26:
For what I'm doing,, 6 - 24 hour greyscale engravings,, I think it's great!!

arkypita, 2019-05-26:
> I set it to run for 1 minute,, pause for 1 minute and I set up a 55 second run and told it to run 10 times. It never stopped to cool. 
> I had one problem - I did a raster and it never paused:

It's strange. Cooling cycle are implemented by sending [feed hold and and cycle resume immediate command](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#ascii-realtime-command-descriptions) and independently from what kind of job (vector/raster etc) is done.

I have tried with raster and it work for me. Can you provide more info?

arkypita, 2019-05-26:
However I have a big doubt.
In old 0.9 or without laser mode the feed hold command does not turn off the laser. So the stop will burn holes where it stops. There is a better command to send for suspend that is compatible with 0.9 and laser?

brakthehun, 2019-05-26:
Mighty only work with 1.1 with $32=1.
If timing didn't need to be perfect,, you could send:
M5
G4P60
M3
In the stream.

The estimated time is way off with the cooling pause.

arkypita, 2019-05-26:
@brakthehun What about `0x84 : Safety Door`

brakthehun, 2019-05-26:
Ok,, so,, My concern is - if you need to walk away,, how do you actually pause it??
If it''s "cooling" from a 0x84,, the only way to get it to resume would be a "resume" (~) command. You hit pause inside the program,, you could cancel the cooling cycle / resume. If you hit the pause button though,, it will already be in a hold state and the program won't know to continue holding.

The program could have a hold type in the auto-cooling menu.
--------------------------------------------

 

 
 You are
 receiving this because you were mentioned.
 it on GitHub, or mute
 the thread.

arkypita, 2019-05-31:
> Mighty only work with 1.1 with $32=1.
> If timing didn't need to be perfect,, you could send:
> M5
> G4P60
> M3
> In the stream.

This require to exactly know if laser is on or off when I decide to insert the pause sequence, beacuse at the end of the pause I should restore the previous laser state.

brakthehun, 2019-06-06:
It's annoying when I forget to turn it off. It would be nice to have a live toggle somewhere.

closed 2019-11-03 by arkypita