arkypita/LaserGRBL #659: New Board Wiring Help Needed

poppen03, 2019-05-20:
I bought a 2 axis Chinese laser engraver that arrived with a bad board. I purchased a new board and  was sent one for a 3 axis machine and no wiring diagram. It already has grbl 0.9.

Can anyone share the connections to use with this board? The x and y axis connections are clear, but which one should I use for the laser (there appear to be 3 2-pin options? Also, There are 2 places to plug in the 12v power supply. Which should I use?  I have a 3w laser that I would like to control with Lasergrbl. 

[image]
[image]
[image]

StuartB4, 2019-05-21:
This is the closest thing I can find, it has a relay on the board instead of the FET (the black thing with F540NS written on it on your board) But I think thats more to do with using the machine as a router and not a laser engraver. Most of it looks similar to your board though.

[image]

poppen03, 2019-05-21:
Thanks! This is close and definitely helps. I'm still puzzled by the (2) 2-pin connections next to the Nano though. I appreciate your help!

StuartB4, 2019-05-21:
Yeah, it's hard to see what the pins are connected to.  I do know that if you have a pair of long nose pliers you can usually grab the little plastic socket around the pins and carefully slide it up off of the pins, that way you might see where the pins go. Maybe they connect to tracks underneath the board though.

StuartB4, 2019-05-21:
The first one that says 12 - 36 must be an input, theres now way to set the voltage between 12 and 36 if it was an output. It may also be connected to the barrel connector on the left as an alternate input.
The second one could be either input or output, need a multimeter to check. May be an output for a fan.

StuartB4, 2019-05-23:
Hi  poppen03.
Just wondered if you are having any luck with it.

poppen03, 2019-05-23:
I should know tomorrow (I have a day off to try it out). In what I consider a minor miracle the seller sent me  this photo with connection notes. I'll post again after I try hooking it up. 
[image]

StuartB4, 2019-05-23:
Ok.
Strange how he marked 2 x 12 volt fan connectors, even though one is clearly marked 12 - 36 volts on the board.
Apart from that it looks pretty straight forward.

poppen03, 2019-05-24:
No joy. No laser control with expected connections (see photo). The motors work fine but no power goes to the laser. 
The board is flashed with grbl 0.9. I moved the jumper over as shown in the above diagram you sent. I did try connecting the laser (2 pin) to the center connections (the 12-36 and the 12 v) and the laser does work, though there is no control over it. 
[image]

StuartB4, 2019-05-24:
Looking at the image you posted, it looks like your laser is wired differently to the socket it's plugged to.
Usually the red wire is + positive, you have it connected to the - negative. Is that the correct way for that laser.

poppen03, 2019-05-24:
Yes! Great catch! Thank you!!!  The laser now comes on and off when the engraving begins and ends. The only thing is that it doesn't seem to be at full power. It's not burning anything as it goes through the program. Any thoughts?

poppen03, 2019-05-24:
Is it better to upgrade to grbl v1.1?

StuartB4, 2019-05-25:
Is the laser focussed correctly, smallest possible dot.
What colour material are you trying to burn, white is always a problem, but that also depends on the lasers wattage and the engraving speed.

Upgrading to GRBL 1.1 is a good idea as it includes extra settings for laser mode.
I would carefully remove the Arduino Nano from the main board to upgrade it, then you should'nt have any problems with interference from it. All you need is the USB lead.

There is another reason why it's not burning, it could be damaged after connecting the laser the wrong way in one socket. Unless it has reverse polarity voltage protection. Usually when laser diodes break they stop LASING but still emit light, more like an LED light.

poppen03, 2019-05-25:
Thanks for those ideas.  The laser is focused and capable of lasing, I'm using a piece of brown cardboard. 
I have found that I can send M3 command directly (through Lasergrbl) and nothing happens, however, if I send M3 S1000 (from Lasergrbl) the laser comes on and starts burning. Still, when I run a file the gcode only sends M3, so it doesn't turn on. Is there somewhere I can set a default to M3 = M3 S1000 or a setting I'm missing?

StuartB4, 2019-05-25:
What speed are you trying to engrave at.
[image]

poppen03, 2019-05-25:
Good question. Where can I find the dialog box in your image? Here's a photo of the current v0.9 config. I will upgrade to v1.1 but I thought I would get it working first.
[image]

poppen03, 2019-05-25:
OK, please disregard what I said. I went ahead and upgraded to 1.1f and it seems to be working. Here is the current 1.1f config. Do you suggest any changes?
[image]

poppen03, 2019-05-25:
I agree that speed may be an issue, since I'm still seeing gaps in burning where faster moves occur. Where is that setting?

poppen03, 2019-05-25:
Here's a photo of my first "successful" burn. Notice the uneven lines. It looks like longer, faster moves don't burn as well.
[image]

StuartB4, 2019-05-25:
> Good question. Where can I find the dialog box in your image?

[image]

Your GRBL settings look fine. Just 2 things though.

1. $30 only needs to be 255.

2. Is the maximum travel of your machine 200 mm. If so then thats fine.
It's only a problem if you use Continuous Jog setting in LaserGRBL settings and the maximum travel
of your machine is less than the setting in GRBL settings $130, $131 and $132.

Continuous jog allows you to hold the mouse button down on any of the jog or Home buttons
and it will continue to jog while the button is down.
As long as your $130, $131 and $132 are set correctly then it will stop jogging when it reaches the limit you set. Or when you release the mouse button.

poppen03, 2019-05-26:
You have been a tremendous help! Thank you, I really appreciate it! It's been a challenge, but now that I have basic functionality I can play around with various settings and find what works best for my projects.

StuartB4, 2019-05-26:
Happy to help if I can.
Good luck with it.
There's always someone here to help if needed.

brakthehun, 2019-06-06:
Have you tried using an M4 command instead of an M3 command??

poppen03, 2019-06-06:
Only M3 works for me.

Testimonial, 2019-10-02:
> No joy. No laser control with expected connections (see photo). The motors work fine but no power goes to the laser.
> The board is flashed with grbl 0.9. I moved the jumper over as shown in the above [...]

Hi, I can see your board is almost same as my just with a small difference and I'd like to figure it out. Can you please help me to identify tiny part located besides 2pin DC 12V laser output please?
[image]

I've got only 3 free holes with label STP1. I know if you read laser on right there is ground, in the middle there is signal up to +5 and left one is stabil 12V.
The name of this part could help me to extend my board.
Thanks a lot.

StuartB4, 2019-10-02:
Looks like a mosfet / transistor.

closed 2019-11-03 by arkypita

usamaijazz, 2021-10-06:
Hi, 
Hope you are doing well
I also Bought the same board form ALiExpress China about 4 years ago as you have mentioned in your first post.

[image]

and just got the time to operate it.
Can you help me out to start working with this board.
If its possible for your then Please send the link for for software to use.
Also any guide video you might know.

ThankYou,

Waiting for your reply.
here is my email id 
usamaijaz1162@gmail.com