arkypita/LaserGRBL #662: Rather slow

TimSmith44, 2019-05-22:
Hi. Laser GRBL. Great program but 5 hours to make a coaster seems a bit slow?

My image is based on line art, and no more than 100 mm x 100 mm. Hardware is Eleksmaker A3 Pro 250 mW and is completely stock. I have asked for 3000 mm / min and 8 lines, one pass.

The same image in Elekscam takes approx 20 minutes but the quality is very poor. Good enough for small logos and bitmaps but not more complex work.

I am not that technical on programming etc so in simple terms, how can I make this run faster on LaserGRBL please?

arkypita, 2019-05-22:
Doing some math your job will need about 26min. (100x8x100/3000).

If not, verify your grbl configuration:
Axis max speed and laser mode enabled.
Assuming you have grbl v1.1 and not the older 0.9

TimSmith44, 2019-05-22:
Thank you. I have improved things some with a less complex image but it still took 2.5 hours so will look into those parameters. Is there an easy way to check the version?
I seem to recall seeing v1.1 written somewhere and its a new purchase.

arkypita, 2019-05-22:
https://user-images.githubusercontent.com/16980026/54475616-309d1b80-47eb-11e9-9fd3-edc5d7500be6.jpg

TimSmith44, 2019-05-22:
Thank you. I will check

TimSmith44, 2019-05-22:
It's GRBL 0.9i  
Can I update it?

TimSmith44, 2019-05-22:
OK I think I answered that for myself and not something I feel comfortable doing really.
Others seem to think that stopped their machines from working.
Are there any settings in 0.9i that are worth tweaking?

arkypita, 2019-05-23:
First of all, you should check speed and acceleration limits in your grbl configuration.
Connect to your machine and open grbl config form in LaserGRBL.

You must set the highest speed value that does not block/stall the motor. Also with regards to acceleration, you should look for the maximum value that allows your marker to respond well.
Values can only be calculated experimentally because depends on hardware.

[image]

The problem you are experiencing does not depend on LaserGRBL, but on the version of grbl on your engraver. Grbl was born as a control firmware for CNC milling machines and since v1.1 grbl does not have specific features for laser.

This means that the "S" command that controls the power is interpreted as "increase in the speed of the tool spindle". Since a spindle has an inertia between when you give it power and when it reaches the actual speed, grbl adds a delay time every time the power is changed.

This delay is not absolutely necessary with a laser, as the laser immediately changes its power since it has no moving parts. Not only that, this undesired effect becomes completely harmful when you use the greyscale line to line marking that changes the power continuously.

Unfortunately there is no way to remove this delay, in my knowledge, in the grbl 0.9 version.
So if you want to keep v0.9 on your board you can only use dithering mode or vectorization.

In grbl v1.1 the grbl developer adds a "laser mode" feature that obviates this problem.

[image]

Contrary to what you claim the switch to grbl 1.1 is not as risky as you think. You can find instructions by searching the web and find support here or from other parties for any difficulty.

My only suggestion before doing this is to take note of the grbl configuration parameters, as you can see them from the dedicated window.

TimSmith44, 2019-05-23:
Thank you so much. My understanding of this has just increased 1,000 % :)
I am in the middle of developing the coasters for sale, so I won't switch right now but very soon.

I found that reducing size to 75 x 75 mm and using Sucki dithering has reduced working time to about 40 to 50 minutes which I can live with. I am very pleased with the results.

Is there any way to get LaserGRBL to start with a central position like Elekscam does?
I see X and Y offsets but not sure how those work as yet.
When burning my wood bowls, it would be very beneficial to have a central start point.

Once again, thank you for your time, and the assist.

arkypita, 2019-05-23:
> I found that reducing size to 75 x 75 mm and using Sucki dithering has reduced working time to about 40 to 50 minutes which I can live with. I am very pleased with the results.

Want to upload a picture of your result?

> Is there any way to get LaserGRBL to start with a central position like Elekscam does?

I have choose that LaserGRBL will use a cartesian axis system with 0,0 in near-left position. This was a choice to make it simple for first time users.

I know that someone can prefer center working, especially who manage round pieces. For this users custom button can solve any kind of customization.

With gcode (hint use G92) and some math you can do whatever you want. 
https://github.com/arkypita/LaserGRBL/issues/306

TimSmith44, 2019-05-23:
Sure, they are here

[image]
[image]

Thanks again. I will try the custom button.

maillemaker, 2019-06-05:
Tim:
There are custom buttons you can add to LaserGRBL to do things like go to center of image, and trace an outline.  If you put your laser into low-power spotting mode (My Eleksmaker A3 Pro has a button on the laser head to do this) the laser will act like a spotter.  Then you click the "center" button and it will go find the center for you.  Or the "trace" button and the machine will trace the boundaries of the image.  These are very useful for locating the work on the piece prior to starting the burn.   You can find the custom buttons in this forum.  They are simple to install.

closed 2019-11-03 by arkypita