arkypita/LaserGRBL #664: Stepper driver DVR8825

peagerfw, 2019-05-23:
Hi,i have new stepper driver dvr8825 but if i engrave 40x40mm size final product is cca 20x20mm
If use stepper driver A4988 engraved size is ok.Please how to set DVR8825?
Thank you.

arkypita, 2019-05-23:
microstepping issue? maybe using google can fix?
https://www.google.it/search&q=A4988+drv8825+microstep

peagerfw, 2019-05-23:
Yes maybe and maybe one normal person give me helpful instruction how to make.

arkypita, 2019-05-23:
and that's what I did, please

peagerfw, 2019-05-23:
I think : open this program,sett this parameter or go to grbl setting use this ....
this is help for me...  but thanks for google search link

StuartB4, 2019-05-23:
I assume you are plugging the 2 different motor drivers in properly.
They have the current adjuster pot at opposite ends when they are plugged in.
Like here:
[image]

These 2 boards would both plug in to the same socket as they are. 
Some people get them wrong and rotate one to match how the other one was in the socket.

peagerfw, 2019-05-23:
yes,i know .. i plugging in properly.
problem is that dvr8825 work different.. with 1/2 stepps 
[image]

StuartB4, 2019-05-24:
You could make sure your motors are calibrated correctly.
Watch this video, you can skip forward to 4:10 in the video.
https://www.youtube.com/watch?v=z3bUg9InVeA

Click this link and download the calibration spreadsheet.
www.triquetra-cnc.com 
When you get there scroll down to the spreadsheet image and click it.
Enter your email address on the right and click "Yes I would like to download the spreadsheet now"

You don't need to keep watching the video, you can just save this image below and read the instructions on it.

[image]

peagerfw, 2019-05-24:
thank you,but this is what i write
[image]
 if adjusted $100, $101 and $102 – [X,Y,Z] steps/mm from 800 to 1600  my stepper motor stop after run and buzzer it.

arkypita, 2019-05-24:
1600 step/mm is a very high value for stepper resolution. Your controller board needs to do a lot of work to generate all the pulses needed to drive stepper at this resolution. For example if you wanna move at 3000mm/min your board should generate 3000/60*1600=80'000 pulse/sec.

Max pulserate for Arduino GRBL is 30khz (30,000 pulses per second).
https://github.com/grbl/grbl/issues/41

consider adjusting the microstepping, as I have already indicated

arkypita, 2019-05-24:
Moreover you don't really need a resolution of 1/1600 of mm.
1600 step/mm is a resolution of 40640 DPI.

Professional printing is done at 300DPI
Configure your microstepping to have something like 200 step/mm or less.

peagerfw, 2019-05-24:
And how configure my microstepping??

arkypita, 2019-05-24:
It's hardware: usually there are some jumper on control board where you put your dvr8825. Usually they are below the driver itself, but not always. If you haven't jumpers your board does not allow changing microstepping.

Example for arduino CNC shield: https://blog.protoneer.co.nz/arduino-cnc-shield-v3-00-assembly-guide/#MicroStep

peagerfw, 2019-05-24:
I have two board Woodpecker and Camtool cnc board where is used dvr8825 and here haven't jumpers.
Is possible hack this board or driver?

arkypita, 2019-05-24:
dvr8825 has 3 pin called M0, M1, M2 that control microstepping.
Based on logic level on those pin (5V high, 0V low) the driver get a different microstepping configuration.
See table here: https://www.pololu.com/product/2132

If you want to hack your board you should follow PCB traces from those pin and cut them, scratch out PCB to reveal copper and solder some wire to +5V or GND to set level on M0 M1 M2

arkypita, 2019-05-24:
Measure those pin with multimeter so you know what is your board microstepping configuration.
You can simply cut traces if you need to put pin to LOW logic level because of internal pull-down.

[image]

brakthehun, 2019-05-25:
Can you post your $$ settings??

peagerfw, 2019-05-25:
$0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=0.000
$30=10000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=1800.000
$111=1800.000
$112=1800.000
$120=100.000
$121=100.000
$122=100.000
$130=300.000
$131=150.000
$132=44.000

brakthehun, 2019-05-26:
Try:
$100=1600
$101=1600

$110=1125
$111=1125

peagerfw, 2019-05-27:
ooooh realy thank you very helpful ,work it ;)
Please, Z axix (102,112) must change too or no?

brakthehun, 2019-05-27:
You can change the Z similarly if you have a Z.

peagerfw, 2019-05-27:
Yes i have ,thank you so much 👍

brakthehun, 2019-06-06:
I think this is solved.

peagerfw, 2019-06-06:
yes

closed 2019-06-11 by peagerfw