arkypita/LaserGRBL #666: Image shifted in +Y axis from origin in software

maillemaker, 2019-05-27:
I have noticed what may be a bug in the LaserGRBL software.

Here is my image:

[image]

I used Paint to move the entire image to the upper-left corner in Paint, and then modified the image boundary to closely fit the image.  The result is above.

When I import this into LaserGRBL, the image is shifted in the +Y direction, like so:

[image]

Sometimes if I click in the graphics area in LaserGRBL the image will suddenly re-center correctly.  Other times, it stays as you see above.

arkypita, 2019-05-27:
Upload a screenshot of this window

[image]

maillemaker, 2019-06-05:
I have been unable to reproduce this error.
I am pretty sure I did not put an offset in the Offset fields, but cannot be sure now.
So, I will close this issue.
Steve

closed 2019-06-05 by maillemaker