arkypita/LaserGRBL #668: my laser runs out of power after 90% of the job

johnbluesboy, 2019-05-30:


StuartB4, 2019-05-30:
Is it always 90% no matter what size image you burn?
Is the laser cooled.

johnbluesboy, 2019-05-31:
Hi yes it seems to be

John Richardson

brakthehun, 2019-06-06:
What laser are you using??
Are you using a diode?? A tube??

johnbluesboy, 2019-06-06:
Hi I am using a 2500w diode it seems to lose power after about a minute of use

John Richardson

brakthehun, 2019-06-06:
Sounds like a faulty diode.
Does the power return after you let it sit??

johnbluesboy, 2019-06-06:
Yes it does after about 30 minutes 

John Richardson

brakthehun, 2019-06-12:
There is anther revision to the software being tested that adds a cool down for diodes. It works very well so far.

johnbluesboy, 2019-06-12:
Hi where can I get the new software?

John Richardson

StuartB4, 2019-06-12:
Here it is.
https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.15

StuartB4, 2019-06-12:
https://github.com/arkypita/LaserGRBL/tree/v3.0.15

arkypita, 2019-06-13:
> Hi where can I get the new software? John Richardson

I had removed it from releases because it works only with grbl v1.1 and produce bad results (burns) with older 0.9-0.8 grbl version. I am working on a version that act better with older version. Just wait for new version.

johnbluesboy, 2019-06-13:
Ok thanks I am using 1.1

John Richardson

closed 2020-01-27 by arkypita