arkypita/LaserGRBL #670: SK language for Czech Republic toooo

00alkskodi00, 2019-06-01:
Hi i have translated the files to Slovak Language, do you need the files?
here is a link: https://mega.nz/#F!q4xyyIjT!KrzBhZ2Yxv7ggl6g6hoS1A

arkypita, 2019-06-03:
[image]

arkypita, 2019-06-03:
file not available

StuartB4, 2019-06-03:
Working for me.
[image]

arkypita, 2019-06-03:
now it works to me too

StuartB4, 2019-06-03:
Ok.

arkypita, 2019-06-03:
@00alkskodi00 

Is your translation sk-SK Slovak (Slovakia) or cs-CZ Czech (Czech Republic)
See: http://www.lingoes.net/en/translator/langcode.htm

peagerfw, 2019-06-26:
Slovak language not work in LaserGRBL v3.0.16

00alkskodi00, 2019-06-27:
Is sk-SK Slovak (Slovakia)

peagerfw, 2019-06-27:
ked prepnem na SK program sa restartne ale jazyk zostane Anglicky

00alkskodi00, 2019-06-27:
Ahoj viem o tom hádam to autor nejako poriesi a možno som urobil chybu ja :)

peagerfw, 2019-06-27:
Uvidime ;) aj ked to mohol preverit skor, ako pustil novu verziu do obehu.

arkypita, 2020-01-15:
Is ok now or not? I don't remember if I fix it

00alkskodi00, 2020-01-15:
Is not a komplet but is working :)