arkypita/LaserGRBL #671: запрос функции для регулировки мощности лазера

serchel111, 2019-06-04:
Здравствуй!
Спасибо за программное обеспечение.
Степень серого на материале зависит от многих причин. И повторяемости гравировки на другом материале ожидать не приходится. Если даже изменили настройки фокуса лазера. В связи с этим хотелось бы иметь утилиту настройки мощности лазера. Я вижу это так:
1. задается предполагаемый уровень мощности лазера "min" на определенной скорости и количество линий на мм. Задается шаг изменения мощности и количество тестовых областей (ниже и выше заданной мощности лазера).
То же самое задается для мощности лазера "мах".
Имеется выбор тест "min" или, и  "max".
2. Отправляется с учетом всех коэффициентов на лазерную гравировку. Гравируется тестовые области размером не менее 5*5мм. (желательно с выгравированными параметрами мощности лазера, но не обязательно).
3. Визуально оценивается оптимальный порог мощности лазера "min", "max"  для данного материала.
Результат сэкономит много времени.
4 . Для полного счастья просится "пипетка" для определения расчетной мощности лазера в интересующих областях растрового изображения. 

5.Ограничение 255 ступеней изменения мощности лазера обусловлено аппаратной поддержкой станка? Тогда можно ли ограничить максимальную мощность лазера? К примеру у меня мощность лазера, при котором начинает темнеть материал (Pmin) = 70. мощность, при котором материал начинает обугливаться (Pmax) = 110. Мой диапазон на материале Pmax-Pmin = 40 ступеней.  Но для меня важны детали , которые находятся в диапазоне, к примеру,  70-80. То есть всего 10 ступеней. 
При обработке растра я делаю диапазон 70-210 и получаю 40 ступеней важного для меня диапазона. Но мощность лазера от (Pmax) = 110 - 210 сжигает мой материал в уголь. Поэтому я хотел бы чтобы Pmax в диапазоне от 110 до 210 равнялось 110.

serchel111, 2019-06-04:
Hello!
Thanks for the software.
The degree of gray on the material depends on many reasons. And the repeatability of engraving on another material is not expected. Even if you change the focus settings of the laser. In this regard, I would like to have a utility to adjust the laser power. I see it like this:

sets the estimated power level of the laser "min" at a certain speed and the number of lines per mm. Sets the step of changing the power and the number of test areas (below and above the specified laser power).
The same is set for laser power "max."
There is a choice test "min" or, and "max".
Sent with all coefficients to laser engraving. Engraved test areas of at least 5 * 5mm. (preferably with engraved laser power parameters, but not necessarily).
Visually estimated optimal threshold power of the laser "min", "max" for this material.
The result will save a lot of time.
four . For complete happiness, a pipette is requested to determine the calculated laser power in the areas of interest in the bitmap image.
5. Limiting the 255 steps of changing the laser power due to the hardware support of the machine? Then is it possible to limit the maximum laser power? For example, I have a laser power at which the material begins to darken (Pmin) = 70. the power at which the material starts to char, (Pmax) = 110. My range on the material is Pmax-Pmin = 40 steps. But for me the details are important, which are in the range, for example, 70-80. That is only 10 steps.
When processing a raster, I make a range of 70-210 and get 40 steps of the range that is important to me. But the laser power from (Pmax) = 110 - 210 burns my material into coal. Therefore, I would like Pmax in the range from 110 to 210 to be 110.c).

arkypita, 2020-01-15:
A "non linear" color to power conversion is already in development roadmap #64

closed 2020-01-15 by arkypita