arkypita/LaserGRBL #677: Enhancement    Custom feed hold and  resume comands

sinfocomp, 2019-06-10:
After a feed hold it woudl be nice to have custom commands (gcodes) to be sent before resuming (just inset them before the next actual gcode command), this would be helpfull like on my case to move the torch to the ignition position and re open the gas & oxigen valves closed on feed hold commands (these would be gcodes sent after the feedhold is instated)
Im using your software for oxi-fuel cutter machine

cristianodems, 2019-06-17:
A subject that would meet this and other cases. It would allow custom buttons to send the hotkeys.

arkypita, 2020-01-15:
Latest version support immediate commands in custom buttons, so you can create a "two state button" for "feed-hold + custom gcode" to stop and with "custom gcode + resume" to start.

closed 2020-01-15 by arkypita