arkypita/LaserGRBL #678: Edit botton

ukadyp, 2019-06-13:
Hi

is there anyway we can add a edit botton in so say if u have loaded the file then gone right to just about laser. i you put the wrong speed in and need to change it we dont have to load up all the file over again. i mean due to sometimes i have laserd on and found i needed to go slower but when loading the file i used the cut part so if you load the file again and try go over u can not get it right due to the cut part has to be right on which you can not do.

StuartB4, 2019-06-13:
What you can do is save the gcode file after you click create in the Target Image settings, but before you laser.
Then you can edit the gcode speed settings.
Just click Save Program on the File menu and give the file a name, and save it.
Then you can open it with a text editor like Notepad or Notepad++ and change the F setting as many times as you like and re-save the file. Then load the GCode file back in with Open File on the File menu. You won't have to change any other settings.

Example.
G90
G0 X0 Y0
M4 S0
F800 (EDIT THIS SETTING FOR THE SPEED)
G0 X18.45 Y2.419 S0
G1 X18.225 S9
X18.113 S10
X18 S11
X17.944 S10
X17.831 S11
X17.775 S13
X17.719 S14
X17.663 S15

ukadyp, 2019-06-13:
thanks i try that. i been doing slate coasters and sometime it not come out right or it needed to be moved abit or edite so been binning so many that not right

closed 2020-01-15 by arkypita