arkypita/LaserGRBL #680: S parameter with vectorize

CrazyCitystars, 2019-06-14:
Hi,

Have a laser that supports PWM.
When doing line to line tracing its no problem.
But when i want to do a cutout it doesn't produce the S parameter at the end of the gcode.

example:

F100
G0 X48.85 Y8.151
M3
G1 X47.45 Y8.112
G1 X46.7 Y8.201
G1 X45.95 Y8.29

While it should be to get it working:

G1 F100
G0 X48.85 Y8.151 S255
G1 X47.45 Y8.112 S255
G1 X46.7 Y8.201 S255
G1 X45.95 Y8.29 S255

Is this a known problem or something i do wrong?

Thank you!

arkypita, 2019-06-14:
S is modal. Once sent does not need to send again unless you want to change.

CrazyCitystars, 2019-06-14:
Yes, but in the first example the S parameter i don't get it at the end of the first command

arkypita, 2019-06-14:
It is on the very first lines

G90
G0 X0 Y0
M5 S255   <=== HERE
F1000
G0 X84.8 Y5.015
M3
G1 X85.65 Y5.003
G1 X87.15 Y5.146
[....]

closed 2020-01-15 by arkypita