arkypita/LaserGRBL #682: Feature request: drawing support delay

NewEmailAddress, 2019-06-15:


NewEmailAddress, 2019-06-15:
Hi,  

I have been using this software for both laser and now with a drawbot. In both cases a delay is needed for some hardware for m3 and m5.  In inkscape there is a plugin (J tech photonics) that adds a delay for m3 but not m5, but for things like flowing ink, you need to raise the pen before moving, and lower it before moving, not just one. Since I don't have the source for the possibly custom version of grbl 0.9 and dealing with the author is problematic, I am looking at it from the other angle.

When vector drawing with fill, there are commands that include the m3 or m5 at the end of something else, as if the order is unimportant for execution. G0 X4324 Y4324 M5.  For my laser hardware even this results in under/overshoot, and it is noticeable especially with ink, as the area is immediately darker in the direction the movement is going, (with laser there are too many other variables).  

My solution is to save the file, open notepad+ and do a fancy search and replace to convert something like G0 X432 Y432 M5 to several lines with a G5 P0.4 in between:

M5
G5 P0.4
G0 X432 Y432

Also for every standalone M3 or M5 I add a similar line for a delay G5 P0.4 (and before the above)

I would be nice to have this a check mark in the options to automatically do this. Also there is redundancy in the appended M3 or M5 commands, so logic to know which is currently active would avoid duplicate delay commands.

Ex.
Y324 M3
Y432 M3

Thanks

galactyus01, 2019-09-26:
se riesci a risolvere questo problema scrivimi per piacere.. ho lo stesso problema e sto cercando soluzione.. io uso una penna al posto del laser e quindi introdurre una pausa sarebbe stupendo.... se trovo qualcosa ti scrivero' qui.. ciao

StuartB4, 2019-09-26:
Ciao galactyus01.
Ho tradotto il tuo commento poiché la maggior parte delle persone qui usa la lingua inglese. In questo modo potresti ricevere più risposte.

Translation of Galactuys01's comment above.

> if you can solve this problem write me please .. I have the same problem and I'm looking for a solution .. I use a pen instead of laser and then introduce a break would be wonderful .... if I find something I'll write you here.

galactyus01, 2019-09-28:
> Ciao,
> 
> Ho usato questo software sia per il laser che ora con un drawbot. In entrambi i casi è necessario un ritardo per alcuni hardware per m3 e m5. In inkscape c'è un plugin (fotonica della tecnologia J) che aggiunge un ritardo per m3 ma non per m5, ma per cose come l'inchiostro che scorre, è necessario alzare la penna prima di spostarla e abbassarla prima di spostarsi, non solo una. Dal momento che non ho la fonte per la versione eventualmente personalizzata di grbl 0.9 e trattare con l'autore è problematico, lo sto guardando dall'altra parte.
> 
> Quando il disegno vettoriale con riempimento, ci sono comandi che includono m3 o m5 alla fine di qualcos'altro, come se l'ordine non fosse importante per l'esecuzione. G0 X4324 Y4324 M5. Per il mio hardware laser anche questo si traduce in under / overshoot, ed è evidente soprattutto con l'inchiostro, poiché l'area è immediatamente più scura nella direzione in cui sta andando il movimento (con il laser ci sono troppe altre variabili).
> 
> La mia soluzione è quella di salvare il file, aprire il blocco note + e fare una ricerca elaborata e sostituirla per convertire qualcosa come G0 X432 Y432 M5 in più righe con un G5 P0.4 tra:
> 
> M5
> G5
> P0.4 G0 X432 Y432
> 
> Inoltre per ogni M3 standalone o M5 aggiungo una linea simile per un ritardo G5 P0.4 (e prima di quanto sopra)
> 
> Sarebbe bello avere questo segno di spunta nelle opzioni per farlo automaticamente. Inoltre c'è ridondanza nei comandi M3 o M5 allegati, quindi la logica di sapere quale è attualmente attivo eviterebbe duplicati dei comandi di ritardo.
> 
> Ex.
> Y324 M3
> Y432 M3
> 
> Grazie

Hello! I haven't forgotten you! if it is useful I have created a simple program. Just upload gcode files ... set what we want to change .. and set what should write instead of that code. .. you press button .. and save the file arranged with pause or other .. you can also modify m3 and m5 with other commands at will .is fast!. thanks
https://we.tl/t-pkIdZfcWeH

closed 2020-01-15 by arkypita