arkypita/LaserGRBL #685: Problem with laser off power

piotrusp, 2019-06-17:
Hi,
Maybe someone will be able to help me with my laser problem. 
I bought a 5W laser with a control module "SW-LD73 V1.1"

https://eu.banggood.com/Wholesale-Warehouse-450nm-5W-Laser-Engraving-Module-Blue-Light-With-TTL-Modulation-wp-Eu-1337358.html?rmmds=search

[image]

I connected this laser module with a 3-pin cable (+, -, PWM) to the shield "GRBL V2.5.02" controlled by Arduino nano V3 to which I've uploaded the firmware grbl 1.1f

[image]

When I power ON the shield, the laser immediately starts with full power and burning big hole. If I do not quickly press the black button on the laser module, then the plywood under the laser begins to burn with fire.  The button on the module switches the laser power to weak, but when I want to engrave something, the laser does not change its power and nothing burns. It start burning when I push the button on the module but do it with full power. 
When I want engrave this pattern:
.  .  .  .  .  .  
he burns something like that
____________
do not stop firing when going from point to point.

StuartB4, 2019-06-17:
Have you enabled PWM mode in LaserGrbl settings.
Have you set $32 to 1 in GRBL Configuration settings.
Also set $30 to 255 and $31 to 0
They can all be done in LaserGRBL.

piotrusp, 2019-06-17:
Today I made the following steps:
1.	Disconected the Arduino nano from the shield
2.	Plug in to my laptop
3.	Open Arduino platform and cleared EEPROM
4.	I’ve uploaded clear firmware “grbl_v1.1f.20170801.hex” by XLoader
5.	Open LaserGRBL 3.0.16
6.	Connected with my Nano
When connected to my nano, I have such GRBL settings:
[image]
[image]
[image]
[image]
and configuration:
$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=0 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=500.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=0 (Laser-mode enable)
$100=250.000 (X-axis travel resolution)
$101=250.000 (Y-axis travel resolution)
$102=250.000 (Z-axis travel resolution)
$110=500.000 (X-axis maximum rate)
$111=500.000 (Y-axis maximum rate)
$112=500.000 (Z-axis maximum rate)
$120=10.000 (X-axis acceleration)
$121=10.000 (Y-axis acceleration)
$122=10.000 (Z-axis acceleration)
$130=200.000 (X-axis maximum travel)
$131=200.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

Then I changed parameters:
$30=1000 (Maximum spindle speed) to $30=255
$32=0 (Laser-mode enable) to $32=1
GRBL said “Config written successfuly!

Then I disconnected from GRBL and checked new settings through the Arduino platform, its OK. Then I unplug Nano from laptop and plug into my Arduino Nano shield.
Then I made following steps:
1.	plug again power cable and USB cable
2.	open LaserGRBL
3.	connect with nano
4.	open file and set M3 to M4
[image]
5.	push button on nano shield and laser start on full power
6.	push button on laser module and laser switch to weak mode
7.	start engraving and stepper motor work well but laser don’t burns
8.	when again pushed button on laser module that burns with too much power
LaserGRBL send properly command but laser don’t burn properly
That I want to engrave:
[image]
This is my result:
[image]

StuartB4, 2019-06-17:
I noticed the Engraving speed in Target image settings is 500, have you tried a higher speed?

piotrusp, 2019-06-17:
changed speed $110,$111,$112 from 500 to 3000
but it did not help anything
then I changed acceleration $113,$114,$115 from 10 to 100
it has increased the overall speed, the plywood does not ignite anymore but still does not stop burning from point to point

StuartB4, 2019-06-17:
You could just be unlucky and bought a laser with faulty laser driver board.

M0USS, 2019-06-17:
Hello,

for the PWM you have to activate

$ 32 = 0 (Laser-mode enable)
by
$ 32 = 1 (Laser-mode enable)

and use the M4 command

piotrusp, 2019-06-17:
Ok, thank you for help.

piotrusp, 2019-06-18:
Is it possible that it has been damaged the arduino control board but not the laser module?

judocka, 2019-09-05:
I am having the same problem on my cnc. I am trying to fix it, but have no progress. Have you have any update? Could you fix it?

piotrusp, 2019-09-07:
Hi,  I have some solutions. Working without grayscale but its enough for
me. Tomorow I sens you some photo.
Piotr

piotrusp, 2019-09-09:
Hi, Im sending some pictures with my shield and laser module. Im not using
PWM pin, only + and GND, and LaserGRBL software with this settings

$0 Step pulse time 3 microseconds Sets time length per step. Minimum 3usec.
$1 Step idle delay 255 milliseconds Sets a short hold delay when stopping
to let dynamics settle before disabling steppers. Value 255 keeps motors
enabled with no delay.
$2 Step pulse invert 0 mask Inverts the step signal. Set axis bit to invert
(00000ZYX).
$3 Step direction invert 0 mask Inverts the direction signal. Set axis bit
to invert (00000ZYX).
$4 Invert step enable pin 0 boolean Inverts the stepper driver enable pin
signal.
$5 Invert limit pins 0 boolean Inverts the all of the limit input pins.
$6 Invert probe pin 0 boolean Inverts the probe input pin signal.
$10 Status report options 1 mask Alters data included in status reports.
$11 Junction deviation 0,010 millimeters Sets how fast Grbl travels through
consecutive motions. Lower value slows it down.
$12 Arc tolerance 0,002 millimeters Sets the G2 and G3 arc tracing accuracy
based on radial error. Beware: A very small value may effect performance.
$13 Report in inches 0 boolean Enables inch units when returning any
position and rate value that is not a settings value.
$20 Soft limits enable 0 boolean Enables soft limits checks within machine
travel and sets alarm when exceeded. Requires homing.
$21 Hard limits enable 0 boolean Enables hard limits. Immediately halts
motion and throws an alarm when switch is triggered.
$22 Homing cycle enable 0 boolean Enables homing cycle. Requires limit
switches on all axes.
$23 Homing direction invert 0 mask Homing searches for a switch in the
positive direction. Set axis bit (00000ZYX) to search in negative direction.
$24 Homing locate feed rate 25,000 mm/min Feed rate to slowly engage limit
switch to determine its location accurately.
$25 Homing search seek rate 500,000 mm/min Seek rate to quickly find the
limit switch before the slower locating phase.
$26 Homing switch debounce delay 250 milliseconds Sets a short delay
between phases of homing cycle to let a switch debounce.
$27 Homing switch pull-off distance 1,000 millimeters Retract distance
after triggering switch to disengage it. Homing will fail if switch isn't
cleared.
$30 Maximum spindle speed 255 RPM Maximum spindle speed. Sets PWM to 100%
duty cycle.
$31 Minimum spindle speed 0 RPM Minimum spindle speed. Sets PWM to 0.4% or
lowest duty cycle.
$32 Laser-mode enable 1 boolean Enables laser mode. Consecutive G1/2/3
commands will not halt when spindle speed is changed.
$100 X-axis travel resolution 100,000 step/mm X-axis travel resolution in
steps per millimeter.
$101 Y-axis travel resolution 100,000 step/mm Y-axis travel resolution in
steps per millimeter.
$102 Z-axis travel resolution 100,000 step/mm Z-axis travel resolution in
steps per millimeter.
$110 X-axis maximum rate 3000,000 mm/min X-axis maximum rate. Used as G0
rapid rate.
$111 Y-axis maximum rate 3000,000 mm/min Y-axis maximum rate. Used as G0
rapid rate.
$112 Z-axis maximum rate 1,000 mm/min Z-axis maximum rate. Used as G0 rapid
rate.
$120 X-axis acceleration 1000,000 mm/sec^2 X-axis acceleration. Used for
motion planning to not exceed motor torque and lose steps.
$121 Y-axis acceleration 1000,000 mm/sec^2 Y-axis acceleration. Used for
motion planning to not exceed motor torque and lose steps.
$122 Z-axis acceleration 1,000 mm/sec^2 Z-axis acceleration. Used for
motion planning to not exceed motor torque and lose steps.
$130 X-axis maximum travel 1000,000 millimeters Maximum X-axis travel
distance from homing switch. Determines valid machine space for soft-limits
and homing search distances.
$131 Y-axis maximum travel 1000,000 millimeters Maximum Y-axis travel
distance from homing switch. Determines valid machine space for soft-limits
and homing search distances.
$132 Z-axis maximum travel 1000,000 millimeters Maximum Z-axis travel
distance from homing switch. Determines valid machine space for soft-limits
and homing search distances.

sob., 7 wrz 2019 o 23:13 Piotr Policht <piotr.policht.pp@gmail.com>
napisał(a):

piotrusp, 2019-10-07:
[image]
[image]
[image]

jeffolemes, 2019-11-16:
Olá gostaria de uma informação! Alguém conseguiu resolver o problema dessa ligação? Parece q não aceita os comandos do grbl. Estou na mesma situação e já verifiquei vários tutoriais e sem resultado satisfatório.

closed 2020-01-15 by arkypita