arkypita/LaserGRBL #687: Jogging doesn't work

kuksi021, 2019-06-22:
Hello Arkypita,
first I want say you thank you  your great work.
I am new user of your software and I am very satisfied.

Unfortunately, I have problem with jogging.

On picture you can see commands but without effects on laser head.

[image]

I use GRBL 1.1z.

Can you help me?

Best regards

arkypita, 2019-06-22:
v1.1z is an unofficial version of grbl. I don't know where it comes from, but it is spread about 2% of LaserGRBL users. Jogging problem was already reported here and also on official grbl board.

https://github.com/arkypita/LaserGRBL/issues/543
https://github.com/gnea/grbl/issues/659

I suggest to flash an official version of grbl in your arduino, but since I don't really know why your seller decide to use a custom version i can't know if maybe there could be some compatibility issue.

kuksi021, 2019-06-22:
I flashed grbl to version 1.1f and now joggin  work.
Thank you very much :)

arkypita, 2019-06-22:
You are welcome

closed 2019-06-22 by arkypita