arkypita/LaserGRBL #688: Constant Reset in LaserGRBL

metalman1004, 2019-06-23:
I have been trying to use LaserGRBL with my new 5500mw laser and it has been giving me difficulties. It can get anywhere from 15 seconds to 45 seconds through an engraving before it gives up and gives an UnexpectedReset error. This is just with vector engraving with no fill. As soon as I apply a fill, the system immediately has an UnexpectedReset Error. Below I'll show one of the last errors that came up, the type of hardware I'm using, and the software. 
I'm also running 1.1 firmware on my control board. 
I am pretty new to all of this, but I'm making progress. Just to note, when I upgraded to my 5500mw laser from my 500mw laser, I started experiencing issues.
[image]
[image]

[image]
[image]
[image]

metalman1004, 2019-06-23:
I have since upgraded my power supply to 6A hoping it will help. It still have issues on vectoring and line to line tracing. It seems to be working fine with dithering though. I would prefer to use vectoring. I don't know if its an issue with the processing capacity of the control board or what.

arkypita, 2019-06-23:
See https://lasergrbl.cim/faq

metalman1004, 2019-06-23:
I usually plug my computer in but now that I'm thinking about it, I have been using battery power for the last 2 days. And I also started using a new USB when I switched my laser. I'll try these solutions tomorrow and touch back!

metalman1004, 2019-06-26:
Well I tried a new USB cable and plugging in my laptop to have a good ground but it is still failing. The program does not even attempt to do filling. It drops right off. So no filling. Pretty frustrating.

arkypita, 2019-06-26:
- How high is your $100, $101, $102?
- Can you post your grbl configuration?
- If you send $C (enter grbl in check mode so no motor movement and no laser action, only streaming) and then you send a job, what is the behaviour?

metalman1004, 2019-06-27:
Here is 100/101/102
$100 | X-axis travel resolution | 80.000 | step/mm
$101 | Y-axis travel resolution | 80.000 | step/mm
$102 | Z-axis travel resolution | 80.000 | step/mm

And here is the full configuration
$0 | Step pulse time | 10 | microseconds
$1 | Step idle delay | 25 | milliseconds
$2 | Step pulse invert | 0 | mask
$3 | Step direction invert | 5 | mask
$4 | Invert step enable pin | 0 | boolean
$5 | Invert limit pins | 0 | boolean
$6 | Invert probe pin | 0 | boolean
$10 | Status report options | 1 | mask
$11 | Junction deviation | 0.010 | millimeters
$12 | Arc tolerance | 0.000 | millimeters
$13 | Report in inches | 0 | boolean
$20 | Soft limits enable | 0 | boolean
$21 | Hard limits enable | 0 | boolean
$22 | Homing cycle enable | 0 | boolean
$23 | Homing direction invert | 0 | mask
$24 | Homing locate feed rate | 25.000 | mm/min
$25 | Homing search seek rate | 500.000 | mm/min
$26 | Homing switch debounce delay | 250 | milliseconds
$27 | Homing switch pull-off distance | 1.000 | millimeters
$30 | Maximum spindle speed | 255 | RPM
$31 | Minimum spindle speed | 0 | RPM
$32 | Laser-mode enable | 1 | boolean
$100 | X-axis travel resolution | 80.000 | step/mm
$101 | Y-axis travel resolution | 80.000 | step/mm
$102 | Z-axis travel resolution | 80.000 | step/mm
$110 | X-axis maximum rate | 5000.000 | mm/min
$111 | Y-axis maximum rate | 5000.000 | mm/min
$112 | Z-axis maximum rate | 4000.000 | mm/min
$120 | X-axis acceleration | 80.000 | mm/sec^2
$121 | Y-axis acceleration | 80.000 | mm/sec^2
$122 | Z-axis acceleration | 80.000 | mm/sec^2
$130 | X-axis maximum travel | 200.000 | millimeters
$131 | Y-axis maximum travel | 200.000 | millimeters
$132 | Z-axis maximum travel | 200.000 | millimeters

I am currently out of the country right now, so I cannot try the check mode test. I will try as soon as I get back on Sunday.

arkypita, 2019-06-29:
Grbl conf is ok.

In my opinion is electrical noise: test it in $C mode or detach arduino (the blue board where you put usb cable) and test it alone, better on different PC. If you still have problems your arduino is failed, in that case try to flash again grbl firmware on it but i'm quite sure you have to buy a new one.

If no communication problem in $C and/or when detached you should work on your hardware grounding/shielding cables insulating motor cable from power cable, adding some capacitor etc.

metalman1004, 2019-06-30:
I tried the test mode and when in test mode, the program runs fine. No errors. I tried running the program with the laser disconnected and the g code runs fine again. I thought maybe my driver board was faulty. I tried a completely different control board. Loaded the Grbl firmware. And ran a vectoring g code with a fill and it functioned totally fine. And ran the whole g code with no errors. The only issue I had was the scaling was wrong. It was supposed to be 70mm by 12mm and it ran it much larger. I need to figure out how to fix the scaling.  I think you were right in that I have a faulty Arduino. I'm going to message the company and ask them to send me another one. I'll run some more test burns tomorrow.

arkypita, 2019-06-30:
$100, 101, 102 configure stepper resolution that directly influence scaling. Do some proportion to find exact value or use stuard excel file (do a search in closed issue)

StuartB4, 2019-06-30:
Hi Metalman1004.
Look at the 8th comment in this thread, it will show you how to use the spreadsheet to calibrate you motors.
https://github.com/arkypita/LaserGRBL/issues/664

closed 2019-08-30 by arkypita