arkypita/LaserGRBL #689: Laser modulation offset

flodu33, 2019-06-25:
Hello every body

I use laser mode, and m4 command.
I know my laser can't work under 10% duty cycle
How can i set my grbl?
I mean, at minimum speed , my signal modulation should be with 10%duty cycle minimum

Thanks

[image]

arkypita, 2019-06-25:
[image]

arkypita, 2019-06-25:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm

arkypita, 2019-06-25:
I am not 100% sure of this, but if you set something different from zero to $31 in grbl you can scale laser power.

arkypita, 2019-06-25:
https://github.com/gnea/grbl/issues/531
https://github.com/gnea/grbl/issues/401
https://github.com/gnea/grbl/issues/49

arkypita, 2019-06-25:
Those are if you want to change this behaviour in grbl.

Alternative way is to change the S values generated by lasergrbl.

[image]

Just set "S-Min" to a value > 0 and LaserGRBL will use this value for pure white.

flodu33, 2019-06-25:
Thanks, i ll try tomorow
Have a good day

Singman33, 2019-08-17:
> Those are if you want to change this behaviour in grbl.
> 
> Alternative way is to change the S values generated by lasergrbl.
> 
> [image]
> 
> Just set "S-Min" to a value > 0 and LaserGRBL will use this value for pure white.

Hi, how fo you get that windows to set parameters ? I never found it !

Singman33, 2019-08-17:
Ok, forget that, you only get it when converting a picture to gcode. That software is missing a lot of things to be usefull, like drawing text or lines without the picture step. :/

arkypita, 2019-08-30:
@Singman33 https://github.com/arkypita/LaserGRBL/issues/85

closed 2019-08-30 by arkypita

AxelGUILLAUMET, 2020-06-22:
Dear friend.

Is the $31 parameter could help to solve this?

https://github.com/grbl/grbl/issues/1637