arkypita/LaserGRBL #691: Forgot how to do maths?

TimSmith44, 2019-06-30:
My LaserGRBL just updated and seems to have forgotten maths.

I have a custom button that with no file loaded works correctly but as soon as a file is loaded, the X and Y works as if there is an offset. (There is none in the file).

Apparently, the centre of a 200 mm square image is now at 175.

I have tried other files and same happens.

Co-ordinates are reading x=0 and y=0 at home position,

Also tried re-starting software, and re-starting laser but the moment a file is loaded my centre is nowhere near centre and running would burn way too high.

StuartB4, 2019-06-30:
Working ok here, version 3.0.17
What are the custom button settings?  can you paste them here so I can give it a go.

TimSmith44, 2019-06-30:
It's not the button Stuart. Been using it for months and with no file loaded it works just fine.
(I can't copy as it's on a different machine but its the normal 3 second centre found on here).
Files that worked fine before the update now act as if they have a built in offset.
The display shows the laser at x = 0 and y = 0 and this is correct physically, yet when asked to centre a 200mm image, (x+200/2) and (y+200/2) it will then centre at x = 175 and y about 176.5 and it shows these figures when the laser travels there for its 3 seconds on low power.

I have a load of files for bowl diameters and it's like the files are ALL wrong now. Very frustrating as its been working just fine since I changed to LaserGRBL a few weeks back

TimSmith44, 2019-06-30:
Button is ...

G0 X[left+200/2] Y[bottom+200/2]
M4 S1
G4 P3
M5
G0 X0 Y0 F3000

StuartB4, 2019-06-30:
Yeah that's strange. You are right about it not being the button, works fine for me.

arkypita, 2019-06-30:
The only changes done between v3.0.16 and v3.0.17 is the support of czech language. I am quite sure that no other stuff are changed. If you want to be sure just download older version

StuartB4, 2019-06-30:
Have you used any software other than LaserGRBL on that computer?
I used Universal Gcode Sender once and forgot to reset all the settings before using LaserGRBL again.
When I did use it the next time, it was still using USG's settings for x and y that were present before I closed the program. Had to go back to it and reset everything before LaserGRBL would work correctly.

TimSmith44, 2019-07-01:
Thanks Guys,

No. That computer does nothing else and it is only since the update but I wonder if maybe Windows has updated or similar? Not sure on that.

I will try going back to v3.0.16. Do I need to remove and re-install?

StuartB4, 2019-07-01:
Not sure if you need to uninstall first. Only ever gone forward. You could also try updating  windows .Net, dont know if it would make a difference but should’nt make it any worse.

arkypita, 2019-07-01:
No need to uninstall. Just download and install older version.
v3.0.16 (June 14) https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.16
v3.0.14 (may 13) https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.14

Older version ask you to update (of course) just skip the request and use it, just to test the issue you describe.

arkypita, 2019-07-01:
However I'm not sure the issue you're reporting is really a bug or just a misunderstanding on your part of how it should work so please share with us your downgrade test result, and if you can try to show us the effects of what you are describing with some screenshoot or video in real usage condition.

TimSmith44, 2019-07-01:
OK I will go shoot some video now.

It's very simple. I have a 200 mm square canvas set up in Photoshop with a logo centred within it. I program the custom button with 200/2 to arrive at a 100mm centre so I can align the 200 mm diameter bowl precisely. Instead of travelling to X100 and Y100 it goes to X175 and Y176.5 (from memory). Even if I replace the maths with 100 instead of 200/2, the result is the same once a file has been loaded.

I will try and get video from both versions and see now I have some time

TimSmith44, 2019-07-01:
OK - I am weirded out by this   lol

I have tried .14 .16 and .17 versions and I have videos but they are taking a while to upload

Images i created in Photoshop, that previously worked just fine, are now exhibiting an 'offset'' for want of a better description.

Images I simply imported form another source work correctly in all 3 versions.

I can easy re-do my images as it seems something has maybe corrupted them somehow?

arkypita, 2019-07-01:
For your information "left" and "bottom" are computer to the real leftmost point of the engraving and to the lowest point of the engraving, computed after the conversion to gcode. They are equivalent to the values reported in correspondence with the dashed lines on the ruler.

Conversion to gcode depends of the setup you insert in "raster import", for example white clip, spot removal, brightness, contrast etc and final image size and offset. So double check your setting.

TimSmith44, 2019-07-01:
Thanks. That's kind of my point. The dashed lines on the ruler are showing exactly correct.

These images are being imported as 1 bit dithering, Stucki, horizontal, and 8 lines per inch

TimSmith44, 2019-07-01:
[image]

[image]

[image]

[image]

TimSmith44, 2019-07-01:
I cannot post a video on here - can I email it?

arkypita, 2019-07-01:
Upload somewere and post link here

TimSmith44, 2019-07-02:
https://youtu.be/QEo6K5wREDI

TimSmith44, 2019-07-06:
Any ideas from the video please?

StuartB4, 2019-07-06:
Hi Tim. Can you upload that image you used in the video to here so I can try it on my machine?
Just click Attach files below and select the image, then when it's done click the Comment button.

TimSmith44, 2019-07-06:
[image]

StuartB4, 2019-07-06:
Here's the result.
Original image loaded.

[image]

Settings.

[image]

Actual image size in X direction.

[image]

Actual image size in Y direction.

[image]

Position of cross using my Custom Center Button is correct.

[image]

Starts engraving at lowest point of the image, bottom right of the H character on the word "Hand" 

[image]

This is my Custom Center Button settings.
[image]

StuartB4, 2019-07-06:
I made another Custom Center Button using the same settings as your's and my machine does exactly the same, goes right off the screen. Just change the button settings to this.

G0 X[left+width/2] Y[bottom+height/2]
M4 S1
G4 P3
M5
G0 X0 Y0 F3000

TimSmith44, 2019-07-07:
Thanks Stuart

All good here now

I thought I was going mad when it happened at first.

closed 2019-07-07 by TimSmith44

StuartB4, 2019-07-07:
No problem. We all get a little mad every now and then.