arkypita/LaserGRBL #692: New smoothiware status format

gerrylenz, 2019-07-01:
Smoothieware used now the status format same as grbl format.
Change in GrblCores.cs line 1475
```javascript
GrblVersionInfo rversion = StatusReportVersion(rline);
if (rversion > new GrblVersionInfo(1, 1))
{
  //grbl > 1.1 - https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports
  string[] arr = rline.Split("|".ToCharArray());

  ParseMachineStatus(arr[0]);

  if (rversion > new GrblVersionInfo(1, 1))
    for (int i = 1; i < arr.Length; i++)
    {
    if (arr[i].StartsWith("Ov:"))
      ParseOverrides(arr[i]);
    else if (arr[i].StartsWith("Bf:"))
      ParseBf(arr[i]);
    else if (arr[i].StartsWith("WPos:"))
      ParseWPos(arr[i]);
    else if (arr[i].StartsWith("MPos:"))
      ParseMPos(arr[i]);
    else if (arr[i].StartsWith("WCO:"))
      ParseWCO(arr[i]);
    else if (arr[i].StartsWith("FS:"))
      ParseFS(arr[i]);
    else if (arr[i].StartsWith("F:"))
      ParseF(arr[i]);
    }
  }
  else //<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
  {
  string[] arr = rline.Split(",".ToCharArray());

  if (arr.Length > 0)
    ParseMachineStatus(arr[0]);
  if (arr.Length > 3)
    SetMPosition(new GPoint(float.Parse(arr[1].Substring(5, arr[1].Length - 5), System.Globalization.NumberFormatInfo.InvariantInfo), float.Parse(arr[2], System.Globalization.NumberFormatInfo.InvariantInfo), float.Parse(arr[3], System.Globalization.NumberFormatInfo.InvariantInfo)));
  if (arr.Length > 6)
    ComputeWCO(new GPoint(float.Parse(arr[4].Substring(5, arr[4].Length - 5), System.Globalization.NumberFormatInfo.InvariantInfo), float.Parse(arr[5], System.Globalization.NumberFormatInfo.InvariantInfo), float.Parse(arr[6], System.Globalization.NumberFormatInfo.InvariantInfo)));
  }

change to
/*
GrblVersionInfo rversion = StatusReportVersion(rline);
if (rversion > new GrblVersionInfo(1, 1))
{
*/
  //grbl > 1.1 - https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports
  string[] arr = rline.Split("|".ToCharArray());

  ParseMachineStatus(arr[0]);

  if (rversion > new GrblVersionInfo(1, 1))
    for (int i = 1; i < arr.Length; i++)
    {
    if (arr[i].StartsWith("Ov:"))
      ParseOverrides(arr[i]);
    else if (arr[i].StartsWith("Bf:"))
      ParseBf(arr[i]);
    else if (arr[i].StartsWith("WPos:"))
      ParseWPos(arr[i]);
    else if (arr[i].StartsWith("MPos:"))
      ParseMPos(arr[i]);
    else if (arr[i].StartsWith("WCO:"))
      ParseWCO(arr[i]);
    else if (arr[i].StartsWith("FS:"))
      ParseFS(arr[i]);
    else if (arr[i].StartsWith("F:"))
      ParseF(arr[i]);
    }
/*    
}
else //<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
{
  string[] arr = rline.Split(",".ToCharArray());

  if (arr.Length > 0)
    ParseMachineStatus(arr[0]);
  if (arr.Length > 3)
    SetMPosition(new GPoint(float.Parse(arr[1].Substring(5, arr[1].Length - 5), System.Globalization.NumberFormatInfo.InvariantInfo), float.Parse(arr[2], System.Globalization.NumberFormatInfo.InvariantInfo), float.Parse(arr[3], System.Globalization.NumberFormatInfo.InvariantInfo)));
  if (arr.Length > 6)
    ComputeWCO(new GPoint(float.Parse(arr[4].Substring(5, arr[4].Length - 5), System.Globalization.NumberFormatInfo.InvariantInfo), float.Parse(arr[5], System.Globalization.NumberFormatInfo.InvariantInfo), float.Parse(arr[6], System.Globalization.NumberFormatInfo.InvariantInfo)));
}
*/
```
Regards
Gerry

arkypita, 2019-08-30:
The "else" is to parse older 0.9 grbl status string, and cannot be commented out.

gerrylenz, 2019-08-30:
correct. Workaround is tested with grbl 1.f