arkypita/LaserGRBL #695: Laser Grbl engraving area.

firewirehn, 2019-07-04:
Hello.I am using 65x50 laser engraving machine with Benbox.Instead of Benbox I instaled LaserGrbl and downloaded one firmware from net to start using instead BenBox.When I uodated firmware my engraving area is only 30x30cm and I can not move machine all along X and Y ahis.Can somebody tell me what to change and how to adjust that my machine traveling all around that I can put homing point on some of corners.

firewirehn, 2019-07-05:
Anyone please

StuartB4, 2019-07-06:
You may need to change the Steps/mm.
To find the correct Steps/mm for your motors look at the 8th post by me in the link below.
https://github.com/arkypita/LaserGRBL/issues/664
When you have the correct numbers follow the steps in the picture below
[image]

firewirehn, 2019-07-17:
Thanks a lot.It helps.

closed 2019-08-29 by arkypita