arkypita/LaserGRBL #697: Laser won't turn on

mrfordnite, 2019-07-07:
I have a 250mw laser, and have followed the tutorial on [instructables.](https://www.instructables.com/id/DIY-Arduino-Mini-Laser-Engraver/) I have everything connected, and my stepper motors are working great. However, my laser is not turning on. Any ideas? Note: I have laser mode enabled. It is also worth noting that the "L" indicator light on my arduino nano lights up on m3 command, and turns off on m4 command.  Here is a [video ](https://drive.google.com/file/d/1-pSiDk878avIIzoZE5EyglDwemtudYOw/view?usp=sharing)of my setup.

StuartB4, 2019-07-07:
Looks like you have the fet transistor next to the laser connector the wrong way round.

mrfordnite, 2019-07-07:
@StuartB4 omg yes i just didn't notice. Will turning it around fix it or will i need to replace the mosfet?

StuartB4, 2019-07-07:
I would remove the laser, turn the fet round and put a multimeter on the laser connector pins to see if there is the correct voltage coming out when you send a laser on command.

StuartB4, 2019-07-08:
Hi mrfordnite.
Did you have any luck with it?

mrfordnite, 2019-07-08:
@Stuart
I do not have the tools to remove the solder. I order a solder pump and
will try tomorrow.

StuartB4, 2019-07-09:
Oh ok. Good luck with it.

StuartB4, 2019-07-18:
Hi. Just wondered if you got it working.

mrfordnite, 2019-08-18:
Hi, @StuartB4, after many backorders and setbacks, I have finally removed and replaced the fet (this time with it facing the right way.) However I am still experiencing a couple of issues...

1. Although the laser now turns on, it is a very weak beam- and it doesn't change when I run an engraving

2. The laser turns off when it should turn on, and is on when it should be off. (M3 turns off, M5 turns on.)

Any suggestions?

closed 2020-01-15 by arkypita