arkypita/LaserGRBL #698: Pausa di lavorazione

distebia, 2019-07-08:
Salve, utilizzo LaserGRBL da quasi un mese e mi trovo benissimo. ho un problema. aprendo un jpg convertito in linea per linea oppure Dithering 1bit BN mi accorgo che, con la lavorazione orizzontale, l'asse x mi crea un doppio margine dovuto all'oscillazione del laser (laser perfettamente serrato) quando completa la linea sia a dx che a sx. potrei utilizzare la marcatura monodirezionale ma il tempo si raddoppia. Allora mi chiedo e vi chiedo: si potrebbe inserire, in fase di conversione, una casella dove scegliere il valore di una pausa dopo ogni linea da riportare nel gcode? grazie

arkypita, 2019-07-09:
Le linee più scure che ti si vengono a formare ai lati dell'immagine sono dovuti alla fase di decelerazione necessaria Peri motori per poter invertire la direzione.

arkypita, 2019-07-09:
Attraverso la modalità M4 attivabile in grbl 1.1 con laser mode attivato, la macchina può effettuare una compensazione che diminuisce la potenza del laser durante le fasi di decelerazione. È sufficiente attivare questa modalità per diminuire molto l'effetto di cui parli.

distebia, 2019-07-09:
Utilizzo grbl 1.1 e modalità laser M4 ma appena l'asse X finisce la linea torna immediatamente indietro creando un doppio margine dell'immagine, sia a destra che a sinistra.

distebia, 2019-07-11:
Ma in fase di conversione jpg non si potrebbe inserire un box dove, eventualmente, impostare un ritardo in ms fra una linea e l'altra? magari non servità a nessuno ma,  come nel mio caso, la macchina oscilla un pò e crea un brutto effetto.

distebia, 2019-07-17:
oggi ho provato a diminuire i valori grbl relativi a velocità e accelerazione per l'asse X ma il problema del doppio bordo usando modalità laser M4 è sempre presente e uguale come prima

closed 2020-01-15 by arkypita