arkypita/LaserGRBL #702: LaserGRBL Picture Output Size

firewirehn, 2019-07-17:
Hello.How to change output size of the picture to desired size.Always is automatically set WxH like aspect ratio.Does anybody know how to set this for example ,I want 100x100 and automatically calculates 100x80.Where to adjust these.

arkypita, 2019-07-17:
LaserGRBL always scale the image proportionally to keep aspect ratio. If you want to stretch/distort it you can edit image outside LaserGRBL (i.e paint, photoshop) to desired size.

firewirehn, 2019-07-17:
Thanks for quick reply.I have one more question How to switch laser before beggining just to know where is start.Need to create custom button for this or not.

StuartB4, 2019-07-17:
This custom button will move the laser at low power around the border of the image you have prepared, so you can position the workpiece where you want.
This is the code I use but you can change the S and F setting numbers to suit you.
S = Laser power level. F = Speed the laser moves around the image border.
You can copy and paste this in to the custom button GCode box.

M4 
S5
F800
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F4000
M5

[image]

StuartB4, 2019-07-17:
This button will find the center of the image you have prepared.
Click once to find center with laser on. Click again to return home and switch laser off.

[image]

GCode box 1.

$32=0
G0 X[left+width/2]
Y[bottom+height/2]
G1 M4 S5 F900

GCode box 2.

$32=1
G0 X0 Y0
M5

firewirehn, 2019-07-17:
Thank you very much it is very helpfull.Thats mean with this custom button i can trace image before burning

firewirehn, 2019-07-25:
Hello.Is there custom button for example to switc on laser only to spot piece of wood to know exactly where laser is.

StuartB4, 2019-07-25:
Try this. Press and release left mouse button to flash laser on and off.
[image]

firewirehn, 2019-07-25:
I tried but nothing happened

StuartB4, 2019-07-25:
Did you change the Button Type to Push Button at the top.
Were you connected at the time.

firewirehn, 2019-07-25:
I did not touch anything actually I don't know how to change that.

StuartB4, 2019-07-25:
Its the drop down menu where it says ButtonType.

firewirehn, 2019-07-25:
I do not have this Button Type..only hotkeys.Can you send me picture to see exactly where is it

firewirehn, 2019-07-25:
For example on the site I founded Toogle Laser on off.There is first Gcode M3 S30 and Second is M5.On mine settings instead M3 I put M4.Shall I put M4 S30...or something eles.Or if you have some example how to set up low light laser 

 
  On čet, jul 25, 2019 at 21:42, spasoje kisic<kisicspasoje@yahoo.com> wrote:   
I do not have this Button Type..only hotkeys.Can you send me picture to see exactly where is it

StuartB4, 2019-07-25:
[image]

[image]

When you've finished click on the Create button.

firewirehn, 2019-08-27:
Hello my friend.I saw your article about connecting laser to cnc board.I ordered 8 wat laser and i have to connect to the cnc board.I am confused about that because I do not know which cable to connect and where.I will send you two picture of laser and cnc board.From laser going two cables one black thick one and red+white connected to white two pin connector.Both of this cable are connected to one cable which end should be connected to power supoly for laser.If you can help me with instruction where and how to connect

StuartB4, 2019-08-27:
Do you have the pictures?

firewirehn, 2019-08-28:
I sent you in attachment.Check message before..i will send again now in attachment

StuartB4, 2019-08-28:
There are no images attached.
Just click "Attach files" at the bottom of the box you type your reply in and select the images.

firewirehn, 2019-08-28:
Check now

StuartB4, 2019-08-28:
Where are you sending them too, all I get is this.
[image]

arkypita, 2019-08-28:
@firewirehn attachment does not work if you reply by mail
all you need to do is to go to https://github.com/arkypita/LaserGRBL/issues/702 and drag-drop images into texbox, or click "Attach files by dragging&dropping".

closed 2020-01-15 by arkypita