arkypita/LaserGRBL #703: LaserGrbl + joypad

distebia, 2019-07-17:
Salve, sono riuscito a far funzionare LaserGrbl mediante joypad usando un programma di emulazione tipo Xpadder. Riesco a fare muovere l'asse X e l'asse Y con la velocità e passi settati su lasergrbl. Per quanto riguarda l'asse Z ho creato 2 pulsanti personalizzati. il limite l'ho incontrato quando vado a vedere cosa potevo personalizzare in Hotkey manager. Non trovo comandi per incrementare la velocità e i passi, comandi per accendere il laser. Non dico tutto ma una buona parte di cosa fa il programma potrebbe essere inserita in hotkey, sopratutto quello che riguarda i movimenti, velocità e step, laser ecc. Tutto questo perchè se devo posizionare la macchina, messa a fuoco mediante abbassamento Z, lo faccio mediante pc e poi vado a verificare se ho agito bene. Se mi metto davanti la macchina con un joypad, magari wireless, agisco all'istante con molta precisione. Sarebbe bello avere tutto questo in mobilità.

arkypita, 2019-08-29:
Si, capisco.
Per l'XY dovrebbe essere già presente tutto il necessario.
Per quanto riguarda la Z mancano, ma ad onor del vero non ho mai avuto pretese di inserire un controllo della Z completo in LaserGRBL.
Incrementare la velocità va bene, ma di quanto? con che passo? fino a quanto?
Accendere e spegnere il laser lo puoi fare con custom-button e i custom button possono essere comandati da hotkey.

closed 2019-08-29 by arkypita

distebia, 2019-08-29:
ok, grazie