arkypita/LaserGRBL #706: Laser don't cut at the beggining of the outline

Epiks38, 2019-07-22:
Hi everyone
I'm cutting detailed stencils, and since a while, the laser dont cut the beggining of the outlines. It misses the 2/3 first millimeters of the outline... But sometimes cut it well... It look like the laser isn't at full power at the beggining: there's the light, but not enough power to cut...
I tried to slow down, clean the lens, reinstall... I can't figure it out.

When i received the machine, it worked perfectly with this soft...

I use laserGRBL on a DIY laser engraver.

StuartB4, 2019-07-22:
Are you using M3 or M4 setting for the laser?

Epiks38, 2019-07-22:
I can only use M3

StuartB4, 2019-07-22:
Oh ok. Is it only the beginning that it happens or does  it ever miss other parts?

Epiks38, 2019-07-23:
Thanks for your help !
Yes its only the beginning, it never misses any other part.

Epiks38, 2019-07-23:
Well it sometimes misses other parts...

StuartB4, 2019-07-24:
Is your machine setup correctly, are the rails clean, is it a belt drive or lead screw machine.
Does the laser move smoothly in X and Y direction.

Grouic, 2019-12-09:
I've got same problem when I cut white paper. The reflexion of laser prevent it to be cutted.
One of the solution is to cut darker paper (not black, but not extrem white) or to add a pause juste after laser acitivation with "G4 P0.5" in your Gcode

closed 2020-01-15 by arkypita