arkypita/LaserGRBL #707: laser engraving starts in 45 degrees

jmennl, 2019-07-28:


StuartB4, 2019-07-28:
What moves wrong? The little blue cross in LaserGRBL or your laser

jmennl, 2019-07-28:
The laser goes the wrong way

StuartB4, 2019-07-28:
Then you must have your motors connected wrong or the settings in grbl are wrong, not LaserGRBL settings.

StuartB4, 2019-07-28:
Have a look at $3 in GRBL configuration here:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

StuartB4, 2019-07-28:
If you are using 1 motor for the X axis and 2 for the Y axis you may have them connected wrong.
You may have the X and 1 of the Y motors connected to the 2 Y axis connectors, that would move the X and 1 of the Y motors when you just want to move the Y.
The other Y Axis motor would then be on it's own on a serepate Y axis connector.

jmennl, 2019-07-30:
Thanks STUARTB4,

I' ve got now the grbl. 0.9i and not the 1.1 and now it works the wright way,
But my hard ware config is like that of the draw robot 2 motors and 1 belt.
But now i gota problem with the output voltage of the ttl output formy laser the power output is 12.3 volts and ttl output is 0.49 low 0.69 volt high this must be 0 or 5 volts for laser on and off.

jmennl, 2019-08-07:
https://1drv.ms/u/s!Aj0D-HJpPw_dlWtdzFG7ymwwWGDe
https://1drv.ms/u/s!Aj0D-HJpPw_dlWoLwqpk_1H5Vlzb

This is my build with a 10w laser and a sainsmart v2 bord and grbl 0.9i 
But know i've got a smal problem with lasergrbl software it works al fine teh software and hardware only grbl 0.9i doesn't support laser  $30 /31/32  and when i use grbl 1.1 i 'LL  have the problem again with 45 degrees startup.

My question is how can i upgrade 0.9i to 1.1 latest version or howto ad the commando's

Regards jmennl

arkypita, 2019-08-29:
https://github.com/arkypita/LaserGRBL/issues/711

closed 2019-08-29 by arkypita