arkypita/LaserGRBL #708: Step/mm issue

Cookies71420, 2019-08-01:
Excuse me for the language i'm french ;)

I have a problem with my laser cnc
I recently  made a DIY laser cutter (7000m/w) all is working fine
i have only a little problem.
I have 3 nema 17 and this board :  https://fr.aliexpress.com/item/32846220735.html
updated to grbl 1.1f 

The steppers are connected to 8mm (1.25mm) threaded shaft

i would work to 1/16 step to improve quality and a little bit faster but it's impossible to set the good value.
i had try to set in 1/8 steps it's done but the cnc is not very fast but the scale is ok
i had try to set full step it's done but de resolution is bad  but the scale is ok

when i try to set 1/16 all works the cnc is faster but the scale is 1/10 mm
if i set 100mm in the model de result is 10mm 
i understand why (the actual step is 256/mm ) but if i put 2560 the steppers dont moves
Any idea ?
best regards

StuartB4, 2019-08-01:
Have a look here:
https://github.com/arkypita/LaserGRBL/issues/664
Scroll down to the 8th post from the top, by me. It may help.

StuartB4, 2019-08-01:
Also, if you want to engrave greyscale images like photo's you will have to change the Laser Driver board to one that has TTL/PWM connections on it. They go from the laser driver board to the controller board you are using. It looks strange being connected to the Fan connector, but after contacting the board manufacturer they confirmed that it is correct.
The Yellow is the TTL/PWM wire, Red is the +ve Power Wire, and the Black is a -ve Ground which has to be connected to the Fan -ve and the TTL -ve on the controller board.
Like this:

[image]

Cookies71420, 2019-08-01:
hi StuartB4 thank's for your reply ;)

For the first thing I had changed the laser connector as you says it's done !

for the steppers what kind of solution should I take ?
I my case if I choose 1/16 the quality is very good and faster but the scale is reduce to 1/10
should I let 1/16 and change the threaded shafts to bigger size (3mm steps) to be faster ?

in fact if I understood what you said about the steppers I can't put the a correct value bigger than 200 step/mm to be powerful ?

[image]

StuartB4, 2019-08-01:
Did you try the spreadsheet calculator ? It's usually pretty accurate for working out the steps/mm.

Cookies71420, 2019-08-01:
> Did you try the spreadsheet calculator ? It's usually pretty accurate for working out the steps/mm.

Yes sure but the as soon I put the good parameters the motors won't start
it only works between 0 to 1200 steps/mm then I put 2560 il fails

StuartB4, 2019-08-01:
Can you upload your GRBL configuration settings to here?

Cookies71420, 2019-08-01:
> Can you upload your GRBL configuration settings to here?

$0=10 (Step pulse time)
$1=255 (Step idle delay)
$2=0 (Step pulse invert)
$3=0 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=3 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=250.000 (Homing locate feed rate)
$25=250.000 (Homing search seek rate)
$26=208 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$100=250.000 (X-axis travel resolution)
$101=250.000 (Y-axis travel resolution)
$102=250.000 (Z-axis travel resolution)
$110=2500.000 (X-axis maximum rate)
$111=2500.000 (Y-axis maximum rate)
$112=2500.000 (Z-axis maximum rate)
$120=90.000 (X-axis acceleration)
$121=90.000 (Y-axis acceleration)
$122=90.000 (Z-axis acceleration)
$130=6000.000 (X-axis maximum travel)
$131=6400.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

Cookies71420, 2019-08-01:
I use this tool
https://blog.prusaprinters.org/calculator/

lead screw M8 pitch 1,25 = 2560 steps/mm

StuartB4, 2019-08-01:
You said 
> "I have 3 nema 17 and this board : https://fr.aliexpress.com/item/32846220735.html
> updated to grbl 1.1f"

But your config file does not show $30, 31 or 32 so you must have an earlier version.

Cookies71420, 2019-08-02:
I don't understand on my coffin I see the parameters
$30=1000
$31=0
$32=1

StuartB4, 2019-08-02:
If you start laserGRBL and click Connect it should show you which version of GRBL you have.
[image]

Then do this.

[image]

You should get this.

[image]

Cookies71420, 2019-08-02:
yes I have the same version 1.1f
$30=1000
$31=0
$32=1

Cookies71420, 2019-08-06:
good evening friends
Problem solved
since it was impossible to parameterize 2560 steps / mm on the motherboard (the stepper motors were cracking), I decided to change the training of these.
indeed I went with a training belt GT2 so the configuration has changed and I went to 40 steps / mm.
Since I can burn at 5000 mm / min and the resolution is better.
Thank you for your advice and if I can help you on another suget it will be with pleasure.

Good night all
cordially.
Cédric

closed 2019-08-06 by Cookies71420

StuartB4, 2019-08-06:
Hi Cédric. 
Happy to see you sorted it out.  :)