arkypita/LaserGRBL #709: GBRL 0.9 card Mosfet damaged

luisrzf, 2019-08-02:
Hi all,

I've this card with a 5500mW laser and I use Inkscape with J Tech Photonics Laser Tool plugin to draw and LaserGBRL to send the data to the device.

I used to configure the laser power from 10 to 100 but last time I selected 150 and after some minutes the laser stopped. I've discovered that the Mosfet in the GBRL card (near the power connector) is very hot and I unplugged the device.

[image]

When I bought the laser and I wired to the GBRL card it was allways on and I contacted with the supplier. They told me I need to connect the laser driver 12V connector to the GBRL card, not directly to the other power supply and using the TTL connector to the GBRL card.

It seems that by this way when I selected more power to the laser I forced the Mosfet to pass more current and is now dead. I'm going to try to replace it and if nothing works again I'll change the GBRL card by other newer one (and newer version).

But if I reach to repair it, how should I wire the laser driver to the GBRL card using each one their power supplies in order to power enough the laser?

Thanks for reading and for your answers in advance.

toromand, 2019-09-06:
Does your laser module has only 2 wire connection? If yes you have two options
- connect as it was and provide a small heatsink / fan to cool down the mosfet 

Option 2 is more complicated 
-buy a TTL driver board for your laser module to replace the original electronics which would  allow you to have a separate 12v feed and a ttl (5v) wire to control the laser. You would also need to find the appropriate arduino pin on your board to drive this wire.

toromand, 2019-09-06:
... And yes, option 3 - buy a more reliable board maybe :)

luisrzf, 2019-09-07:
Hi toromand,

Thanks for your answer. I've replaced the MOSFET but the problem is still there.
I've bought another better card (I hope).

I'll try with this new one and I'll consider your suggestion about the laser driver.

closed 2020-01-15 by arkypita