arkypita/LaserGRBL #710: Laser will not turn on, switched from BenBox to GRBL

junior1882, 2019-08-08:
Let me first apologize as I am a complete noob to this.  I ordered a cheap laser engraver to get my feet wet if you will.  I had issues with the software the setup came with (BenBox).  I flashed(?) the board so that I could use GRBL no the setup will connect however the laser will not turn on nor can I toggle the laser in any direction.  Any help will with this will e greatly appreciated.  Again I am a newcomer to this.

bafconsulting, 2019-08-10:
mine wont turn on either. Im using CNC 3018 pro newest hardware. Laser does not work.

WHAAWHOO, 2019-08-10:
In the same boat also, any help yet

toromand, 2019-09-06:
Benbox and GRBL 1.1 use different arduino pin to control the laser. (D12 and D11 if I recall well). Some controller boards had a jumper (my old L7 has one under the Nano)

closed 2020-01-15 by arkypita