arkypita/LaserGRBL #711: How to upgrade grbl 0.9i to latest version 1.1

jmennl, 2019-08-09:
This is my new project and need some help with it. 

https://1drv.ms/u/s!Aj0D-HJpPw_dlWtdzFG7ymwwWGDe 

https://1drv.ms/u/s!Aj0D-HJpPw_dlWoLwqpk_1H5Vlzb 

This is my build with a 10w laser and a sainsmart v2 bord and grbl 0.9i 
But know i've got a smal problem with lasergrbl software it works al fine the software and hardware  
only grbl 0.9i doesn't support laser $30 /31/32 and when i use grbl 1.1 i 'LL have the problem again with 45 degrees startup.My question is how can i upgrade 0.9i to 1.1 latest version or how to ad the commands

Regards jmennl
--

arkypita, 2019-08-29:
If you use a CoreXY machine you should configure and compile grbl 1.1 for CoreXY, that is not the default firmware configuration. Read or ask about it on grbl page, not here: https://github.com/gnea/grbl/issues?utf8=✓&q=is%3Aissue+corexy

closed 2019-08-29 by arkypita

juansr87, 2020-03-05:
Excuse me can you pass me the hex 0.9i I can't find it anywhere please