arkypita/LaserGRBL #713: Benbox

WHAAWHOO, 2019-08-10:
How to get rid of benbox. Can I get another controller board like I have on my 3018pro

StuartB4, 2019-08-11:
You could get one of these. CNC Spindle and Laser controller in one.
https://www.sainsmart.com/collections/cnc/products/3-axis-grbl-usb-driver-controller-board

WHAAWHOO, 2019-08-12:
What is the difference between the one you listed and this one that I
bought. And what has to be done to make it work properly.  I'm new to this

Dpsspower, 2019-08-13:
Sorry guys but this is NOT a LaserGRBL issue!

WHAAWHOO, 2019-08-14:
But if this benbox was so great folks would not be talking laserGRBL on
here, right. I'm trying to fix it one way or another

WHAAWHOO, 2019-09-15:
> You could get one of these. CNC Spindle and Laser controller in one.
> https://www.sainsmart.com/collections/cnc/products/3-axis-grbl-usb-driver-controller-board

If I get one of these boards what else would I need to do.  Is it as simple as replacing the board and then using the LaserGRBL that I have already.  Im still trying to learn

StuartB4, 2019-09-16:
You should be able to, depending on the connectors on your stepper motors, power supply voltage (this one takes 12volt supply) and laser connections.
Do you have any images of the machine you want to put it on?

WHAAWHOO, 2019-09-16:
file:///home/chronos/u-d25ae8eb34ffb42b2760173dba0bc0d709e64b3a/MyFiles/Downloads/50X65cm%202500MW%20DIY%20Desktop%20Laser%20Engraving%20Machine%20Cutter%20Printer%20Carver%20US%20606989577429%20_%20eBay.mhtml

Had rough day at work,machine is in garage, but the link above is the
machine I have except only 2 stepper motor.  Your help is greatly
appreciated. I also have a 3018 pro that has laser and a carving spindle,
all the connections look the same as that

StuartB4, 2019-09-17:
Cant see the picture, wrong link.
You mean like this one ?
https://www.ebay.co.uk/itm/NO-VAT-50X65cm-2500MW-DIY-Desktop-Laser-Engraving-Machine-Cutter-Printer-Carver/222581499786?epid=17005353505&hash=item33d2e4278a:g:RQAAAOSw1pZdOsGC

WHAAWHOO, 2019-09-17:
Yes that looks just mine

StuartB4, 2019-09-17:
Should be ok then, looks like pretty standard connections.

WHAAWHOO, 2019-09-20:
I received the board today and when I went to connect the laser it would not connect, the board is a two pin connection and my laser is a 3 pin

StuartB4, 2019-09-20:
**Did you get this board?**

[image]

**If you did then you connect the laser like this.
The yellow wire in the picture is the PWM/TTL positive, it may be Blue or White, Usually they are yellow.**

[image]

**I asked Sainsmart who make the board and this is the reply.**

> Hi Stuart,

Since the FAN interface was connected to the 12V power supply directly, the positive and negative terminals of the laser module should be connected to the FAN. That is, the red (positive 12 volt) should be connected to the FAN+ port of the control board.

The Black (Negative/Gnd) should be connected to the FAN- and TTL- port because the TTL- and GND are not in the same GND and they should be connected.

If you used the laser port as the power source, the laser module will not work because the laser module needs the continuous power supply to control the laser intensity 

> .

WHAAWHOO, 2019-09-21:
Yes that is the board, I read somewhere about the same thing you just to
do.  Thanks

StuartB4, 2019-09-21:
Ok, good luck with it.

WHAAWHOO, 2019-09-22:
Thanks for all the help, looks like its working now, had to fabricate a
connection without splicing any wires, since I had plenty of  connectors
from led lights and some other projects. Thanks again

StuartB4, 2019-09-22:
Thats good to know. If you need any help anytime the LaserGRBL facebook page is a good source.
https://www.facebook.com/groups/486886768471991

WHAAWHOO, 2019-09-22:
Thanks for that info, maybe will there right now, just checked on my test
project and it is doing it mirrored.
Thanks again

StuartB4, 2019-09-22:
Have a look here:
https://github.com/arkypita/LaserGRBL/issues/725

WHAAWHOO, 2019-09-22:
Actually on that page now

WHAAWHOO, 2019-09-22:
Got it I think for some reason in the config file  X and Y got changed to
inverted, just changed to back to 0 and it seems to be working now

StuartB4, 2019-09-22:
Good news. :+1:

closed 2020-01-15 by arkypita