arkypita/LaserGRBL #714: I bought a similar one and when trying to run it from LaserGRBL I had problems. With grbl firmware the motors became hot and it didn't run well, then when tried to put original firmware to run again the chinese software benbox i didn't find the firmware on the net and the seller told me it wasn't opensource.

WHAAWHOO, 2019-08-12:
I bought a similar one and when trying to run it from LaserGRBL I had problems. With grbl firmware the motors became hot and it didn't run well, then when tried to put original firmware to run again the chinese software benbox i didn't find the firmware on the net and the seller told me it wasn't opensource.
I finally bought an arduino UNO with cnc shield, that's a lot better; the old shield neither had PWM.

_Originally posted by @gisep in https://github.com/arkypita/LaserGRBL/issues/436#issuecomment-418770935_

WHAAWHOO, 2019-08-12:
what is involved in switching out the boards I have this board a camtool cnc v3.3d.

Dpsspower, 2019-08-13:
What do you want to say????

closed 2020-01-15 by arkypita