arkypita/LaserGRBL #715: Communication problem Mega2560 with Mega-GRBL

Dpsspower, 2019-08-13:
I have a communication problem with Mega-GRBL.
The board with firmware Mega-GRBL 1f works fine with all terminal programs and tools, but LaserGRBL cannot make the connection.
I have to restart the board and / or the software many times until I get a connect.
This is the same at different versions of LaserGRBL and different PCs and also different versions of Windows (10, 7, XP).

I used a port monitor tool now and it seems that laserGRBL does not read the starting string "Grbl 1.1f ['$' for help]".
Instead of that, it starts sending 13 HEX followed by infinite "?".
The read request size IRP_MJ_READ is just set to 2 bytes which is much too less for the startting string.
I tested terminal programs like TERATERM, which use a 1024 byte IRP_MJ_READ. There the starting strings instantly is received and there is no connection problem.
It seems for me that the starting string somewhere is hanging in the communication pipe and LaserGRBL does not read it.
Instead of that it maybe is waiting for an "ok", which is never coming (or hanging in the pipe after the starting string).

I read about more communication problems with Mega-GRBL and LaserGRBL. But the people mostly think that it is a port setting problem.
I think the problem is when setting up the communication.
LaserGRBL is the only program I tested which starts sending 18H followed by "? "? "?" immediately without waiting for the starting string.
Another thought is, that the Mega2560 board is too fast at startup, so LaserGRBL misses the string (?).
I don't have a solution until now, but trying to connect Mega-GRBL to this nice Laser software is a pain in the ass.

Any suggestions?

regards
Joachim

Dpsspower, 2019-08-20:
Nobody? :-(

arkypita, 2019-08-29:
I am sorry but I cannot test the issue you report 'cause I dont'have an arduino mega board.
Did you try changing some settings in LaserGRBL like soft-reset, hard-reset and threading mode?

[image]

arkypita, 2019-08-29:
> LaserGRBL is the only program I tested which starts sending 18H followed by "? "? "?" immediately without waiting for the starting string.

When the connection is opened LaserGRBL goes in connecting state and a timer is started.

```
SetStatus(MacStatus.Connecting);
connectStart = Tools.HiResTimer.TotalMilliseconds;
```

Now if LaserGRBL receive any status report from grbl that report the machine state (Idle, Run, Hold, Door...) the status shown in statusbar lower-right reflect the status received, but if it does not receive a status that change from Connecting to something else in less then 10 seconds, a timeout expire and the connection will be closed.

```
if (MachineStatus == MacStatus.Connecting && Tools.HiResTimer.TotalMilliseconds - connectStart > 10000)
	OnConnectTimeout();
```

Sending "?" is the only way LaserGRBL has to force grbl to reply with status report.

arkypita, 2019-08-29:
LaserGRBL contains a diagnostic mode can be set by command prompt that log the whole communication process.

If you start LaserGRBL from command prompt or by a shortcut with "comlog" argument "C:\Program Files (x86)\LaserGRBL\LaserGRBL.exe comlog" the program will record a file called "comlog.txt" located in "C:\Users\YourName\AppData\Roaming\LaserGRBL".

In comlog file you can find all the serial data transmitted/received from LaserGRBL.
Here is how a valid comlog look like: https://github.com/arkypita/LaserGRBL/files/2636346/comlog.txt

Dpsspower, 2019-08-29:
After trying hard/soft reset settings without result, I found out that setting Threading mode to SLOW makes chance of connection better. But not perfect.
It needs up to 5 tries "connect - disconnect" until there is a response.
Sometimes, the software also crashes, when doing too many tries.
So this seems to be an unsafe workaround.
Unfortunately I don't understand completely the startup mechanism in the software and also don't have the programming setup to recompile the software.
I can see that you send out "." and then "?" repetively. And you wait for the status message of GRBL. But I cannot find, where the welcome string of GRBL is processed.

My guess is, that the software does not always proper read the welcome string from GRBL (version) or does not clear the RX-pipe after connecting.
The hardware sends the welcome string without any request from a host just after reset. This is in my eyes a very bad thing of GRBL. No data should be sent without request in a proper communication.
I am currenty disabling the startup string in GRBL and then will try if I can get communication to LaserGRBL more reliable.
The sad thing for this software is, that all other programs I tested get instant connection to communicate with the board. This indicates no board issue for me.
Naturally the small Arduino Nano board with GRBL is not as fast like the Mega board. This could explain, why there are less problems with the Nano.

Dpsspower, 2019-08-29:
Oh, you were faster then me. I will try the log.

arkypita, 2019-08-29:
> My guess is, that the software does not always proper read the welcome string from GRBL (version) or does not clear the RX-pipe after connecting.
> The hardware sends the welcome string without any request from a host just after reset. This is in my eyes a very bad thing of GRBL. 

LaserGRBL does not needs the "Grbl 1.1f ['$' for help]" welcome string to consider itself connected.

LaserGRBL needs to correctly parse a status report message like "Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:S>" to consider itself connected.

IMHO your mega board is not replying to "?" with a status report, or it is replying with a malformed (or out of specific) one.

arkypita, 2019-08-29:
> The sad thing for this software is, that all other programs I tested get instant connection to communicate with the board. This indicates no board issue for me.

Other programs does not use data from status report at all, and they consider themselves connected just because they successfully open the com. As side effect they usually does not use the status (Idle, Run, Hold, Jog, Alarm, Door, Check, Home, Sleep) to filter/condition their UI button to be active/enabled only at the right condition, giving the user the task of knowing when it is right or not right to press the buttons.

LaserGRBL only enable buttons when it know it is right to press them, but it needs status report to do that.

arkypita, 2019-08-29:
If you upload comlog.txt also upload sessionlog.txt

sinfocomp, 2019-11-27:
i have the same problem, and the way i manage to live with it is to click connect, wait for timeout wait 10 secconds, hit connect and it works, if you just keep trying to connect it never does untill you wait some time between the firstand seccond connect attempt, i see noine followed, i will do the commlog and upload, thanks

sinfocomp, 2019-11-27:
Hello, attached are the files, hope this helps connect lasergrbl to my Mega cleanly
thanks
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/3897463/comlog.txt)
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/3897464/sessionlog.txt)

arkypita, 2019-11-27:
Hi @sinfocomp 
As I already said, I cannot test the issue you report because I dont'have an arduino mega board. Among other things it is quite probable that with a different card the problem does not occur.

However, thanks for submit your file.

comlog.txt records all the communication between LaserGRBL and grbl, and it is clearly showing what happens. First column is time from LaserGRBL startup in milliseconds, second column is a progressive counter, third column is the kind of action (tx data or rx data, or com port action like open, close etc), fourth column is data (read or written).

- LaserGRBL open the com port from PC side. (first line, 6.6s after startup)
- Then it send the real time command 0x18 (ctrl-x) that should cause grbl to reset itself and send the welcome message. (second line)
- Then it sends 0x3F command each 500mSec (Status Report Query) to encourage grbl to send him the status line.

For 10 seconds he does this job, but receives no response from grbl: no welcome message and no status report, so after 10 seconds it close the connection (at line 22, 16.6s after startup, 10s after connect).

After few second you press again the connect button and LaserGRBL open the com port (line 23, second 28.5 after startup). LaserGRBL does exactly the same things (0x18, than start sending 0x3F every  500mSec) but now grbl react correctly sending welcome message (Grbl 1.1l ['$' for help]\r @ line 27) and sending status report each time LaserGRBL send 0x3F.

```
00006644	00000	com	Open COM17 @ 115200 baud Ctrl-X
00006704	00001	tx	[18]
00006712	00002	tx	[3F]
00007220	00003	tx	[3F]
00007716	00004	tx	[3F]
00008217	00005	tx	[3F]
00008718	00006	tx	[3F]
00009219	00007	tx	[3F]
00009716	00008	tx	[3F]
00010217	00009	tx	[3F]
00010718	00010	tx	[3F]
00011218	00011	tx	[3F]
00011720	00012	tx	[3F]
00012218	00013	tx	[3F]
00012718	00014	tx	[3F]
00013215	00015	tx	[3F]
00013717	00016	tx	[3F]
00014216	00017	tx	[3F]
00014719	00018	tx	[3F]
00015218	00019	tx	[3F]
00015716	00020	tx	[3F]
00016216	00021	tx	[3F]
00016635	00022	com	Close COM17 [CORE]
00028574	00023	com	Open COM17 @ 115200 baud Ctrl-X
00028589	00024	tx	[18]
00028593	00025	tx	[3F]
00028595	00026	rx	\r
00028597	00027	rx	Grbl 1.1l ['$' for help]\r
00028648	00028	rx	<Idle|WPos:0.000,0.000,0.000|Bf:35,255|FS:0,0|Pn:ZS|WCO:0.000,0.000,0.000>\r
00029116	00029	tx	[3F]
00029125	00030	rx	<Idle|WPos:0.000,0.000,0.000|Bf:35,255|FS:0,0|Pn:ZS|Ov:100,100,100>\r
00029614	00031	tx	[3F]
00029622	00032	rx	<Idle|WPos:0.000,0.000,0.000|Bf:35,255|FS:0,0|Pn:ZS>\r
00030116	00033	tx	[3F]
00030124	00034	rx	<Idle|WPos:0.000,0.000,0.000|Bf:35,255|FS:0,0|Pn:ZS>\r
00030613	00035	tx	[3F]
00030620	00036	rx	<Idle|WPos:0.000,0.000,0.000|Bf:35,255|FS:0,0|Pn:ZS>\r
00031113	00037	tx	[3F]
00031122	00038	rx	<Idle|WPos:0.000,0.000,0.000|Bf:35,255|FS:0,0|Pn:ZS>\r
00031591	00039	com	Close COM17 [USER]
```

All this leads me to say that LaserGRBL does exactly what it should do for a correct communication.

I don't know why grbl does not react correctly.
I am the developer of LaserGRBL, not the developer of grbl. 

In my opinion you should forward the question to the developers of grbl-mega firmware.

But few question for you.

- did you already try to uncheck "soft reset" in settings page?
- did you already try to check "hard reset" in same page?
- and all combination of them?

[image]

Let me know

arkypita, 2019-11-27:
I read Grbl version **1.1l**
Where did your firmware come from?

sinfocomp, 2019-12-09:
Hello, thanks for your response, yes i tryed with all soft/hard reset conbinations same problem, the most weird part is that neither UGS, candle, and other guis i have tryed have the same problem, the firmware comes from https://github.com/fra589/grbl-Mega-5X will contact him and see what happens

arkypita, 2019-12-09:
If you want you can dump (spy) a communication session between UGS (and/or candle) and your board, as well between LaserGRBL and your board. 

There is a software that can do that for you, called "serial port monitor".
https://www.eltima.com/products/serial-port-monitor/

Download, install, select menu "session - new session", select the com port you want to spy, check all options excluding modbus, and start monitoring.

[image]

Then open UGS, connect to the board, do some thing (like sending a simple command like G0 X10 Y10) and finally disconnect.

Go back to serial port monitor and select "session - close session", save the session with a filename like "UGS.spm".

Then do the same for LaserGRBL (but please configure LaserGRBL with this settings).
[image]

Then zip both .spm files (ugs.spm and lasergrbl.spm, and if you want candle too) and upload here the .zip file.

3RD-Dimension, 2019-12-17:
Hmmm something seems strange with your Mega firmware... I just tried LaserGRBL with my Mega with a fresh copy of GRBL Mega version and it connects straight away with no problems. I thought it might be a one off so connected and reconnected 10 times and its still ok.

I am using GRBL-Mega 1.1g though - maybe something with the "I" version is causing it?

Dpsspower, 2019-12-17:
Then there must be also something strange with my Mega firmware, because I have the same problem since one year.
I use Mega GRBL 1.1f https://github.com/gnea/grbl-Mega/releases/tag/v1.1f.20170802
What is version 1.1g and where did you get it from?
I cannot find any version "g". With my firmware, "f" is the latest release.

closed 2020-01-15 by arkypita