arkypita/LaserGRBL #716: How to change microstepping for DRV8825?

Mattthev, 2019-08-13:
I'm very new in CNC and this whole stuff. I had the same issue as @peagerfw https://github.com/arkypita/LaserGRBL/issues/664 and changing steps to 1600 and limit max speed to 1125 actually work, but the work is slow for my needs. If I'm correct, lowering microsteps from 1/16 to 1/8 or 1/4 should help me to achieve more mm/min speed, right?
I have Woodpecker board V3.3 and there are no jumpers, I tried to Google some advices, but I'm still confused...

> The resolution (step size) selector inputs (MODE0, MODE1, and MODE2) enable selection from the six step resolutions according to the table below. All three selector inputs have internal 100kΩ pull-down resistors, so leaving these three microstep selection pins disconnected results in full-step mode.

According to this config table I should somehow connect M0 and M1 instead of M2 to get 1/8, right?
[image]
My drv8825 has all the pins soldered like in this picture so I need to unsolder it on the Woodpecker board instead on the driver?
[image]

peagerfw in his issue also mentioned that A4988 is working with $100 = 800 it should be 1/8 microsteps then. So I've ordered on Ali these drivers If I wouldn't be able to change microsteps for drv8825, might there be any issue? Or are these drivers just plug and play?

Thank you and sorry for probably stupid questions, but I can't seem to find good explanations.

tomturican, 2019-08-15:
Hallo,

see this manual on https://www.pololu.com/product/2133 .
I didn´t know how this 3 pins are connected on the woodpecker board. Seems, that all 3 pins are on logic high (5V). You can find it out with a voltmeter.
You are right, for 1/8 step the Mode 2 pin should be open or logic low (0V) and the Mode 0 and 1 pin on logic high (5V).
Maybe the board has some solderbridges instead of jumpers? In this case you can simple open this connection.
But why is this board so slow? I have build my own board and i work with speeds up to 12.000 mm/min in 1/32 without problems. But it is important, the the motor current is carefully adjusted and the power source strong enough, to drive this.

Mattthev, 2019-08-18:
Tomorrow I will explore the board more. I've read that GRBL with Arduino is limited to 30kHZ so it's 1600/60*1125 I've tried to adjust max steps and I'm running it now at 1250 mm/min without any problem. Maybe the limit is little bit higher for GRBL 1.1 or maybe it is due to newer Arduino chip?
I'm using the original 24V 5A power supply. I need to measure the power at wall, but the Ali seller said, that board is limited to 150W and 5A. It should fit 36V power supply also, so I don't know how they come up with 150W... They have 300W board, but only in China, I've ordered CNC from Spain stock. I'm thinking of buying more powerful laser too, I have 5500mW. I wanted 7W, but again only in China stock...

tomturican, 2019-08-18:
Hm. The laserpower.... I have 3 Lasers rated with 2.5W from china. All of them was original equiped with (very very) bad drivers and all of them was adjusted to 1.5W or less... Now i drive them with 1,6A (LD current) and my laserpowermeter shows me about 2.3W. Then i bought a 5W laser.... my lpm showed me 3W.... The laserdiode current was set to 2A. I tried to rise the ld current to 3A and in result the driver burned and the ld was dead.
Now i use a selfmade laser with NUBM08 Laserdiode and a selfmade analog laserdriver. Current set to 3A according datasheet and my lpm shows me 4.7 watt. This laser is strong enought for me but after 10 mins at max power a cooling cycle is needed. ..... ok this was a short story ;-)

What i want to say is.... If you have a spindle driven machine, than the speed is limited. It is not only the max frequency of the arduino even the stepper motors goes close to the max rpm. A spindle driven machine is more usefull for stronger mechanical use like engraving spindles. And in this case is a more powerfull laser not the first option because you can´t use the speed benefit.
... and you should check the real power from your laser. A LPM is absolut yousefull! In my case i ordered 4 lasers and all of them was bad calibrated or wrong declared.

Mattthev, 2019-08-18:
Yeah, I don't think it is real 5500mW... I've read that is mostly half of the power. I can try measure the current on the 3 pin. Yes, I have spindle driven machine 3018 Pro. I was thinking of engraving stainless steel so I need more powerful laser, like 15W? I found some with good reviews on Ali.
This 5500mW is good enough for me for now, I can engrave logo and text on most hardwoods under 10 minutes, but 5 minutes would be better anyway :D I use speed around 900-1000mm/min for hardwood.
Also do you have tip for better laser focus? It seems I'm not able to focus it right everytime... Since I can move the Z axis is there any advantage for the laser power to be closer or further from the wood? My laser point is not a dot but actually a small line... Thank you.

tomturican, 2019-08-18:
STOP! Don´t do that! Don´t do anything with your laserdiode, if you are not sure what you are doing!
Lasediodes are one of the most sensitive electronic parts. A simple measurement an the ld can destroy this part. The best and safest way is the measurement of the current with a suitable clamp meter (contactless). But first you need a datasheet for the mounted ld. Please never change any settings before you know, how high the current should be.

tomturican, 2019-08-18:
The Focus... It is normal that this is a small line because this is the form of the lasercrystal. You should adjust the focus on the laser itself. With standart 3 elements lens (most lasers are selled with this) the distance to the surface should be aprox 7-10cm. Then adjust the laser and you should get a dot with 0,1-0,2 mm (lower the laserpower to the minimum, use a dark surface and use the protection glaces, then you can see it exactly).
With my (real) 2,5W laser i can engrave wood, depending of the type, with speeds between 1500 und 3000 mm/min.

Mattthev, 2019-08-19:
You have very nice laser CNC, the laser seems a lot more powerful than mine. Thanks for the focus tips. I think I found laser issue. When move the Z axis up to get the 7 cm distance and tried focusing the laser, I saw that there are actually two dots... one is the line and second one is just small dot. It wasn't that visible from closer distance.

[image]
I guess it is not right? I will open a dispute on Ali and try to get new laser or at least money for new laser. I'Ve also measured power draw at wall on full laser power it is only 27W and X axis was moving also.
What do you say on this one? https://www.aliexpress.com/item/33022812797.html
It has good reviews on most sellers and real wattage should be like 3.5W. Or any tips for good laser around 100 USD? Thank you.

tomturican, 2019-08-19:
2 dots.... a little bit of spreading is normal for cheap lasers. I think, the manufacturers will premilary use B-Type diodes. But it seems that this is not right focused. My dot size is aprox 0,2mm. Have your found the smallest possible dot size? Maybe you need a highter distance... A small dot size is essential for the burning power....

Mattthev, 2019-08-19:
I took the photo before complete focus probably. I can't go much more higher only about 0.5cm more.
I will open the dispute and see if I can get any money back, since it is not 5W and I don't like the second dot, I haven't found on Youtube any other laser with this kind of issue... 
It is not 15W for sure but I like the 12V and 5A too, I found review when they mentioned 3.5W and another said 5W. Not sure if those review are for the same laser but images are the same. This seller has it with EU plug, the other ones mostly only US plug. Also some of the other sellers mentioned that it is for metal engraving too. If the laser would be at least like yours, I will be probably happy enough :D

tomturican, 2019-08-19:
;-) Then i wish you good luck. I think the price is not too high to test it out. ;-)

closed 2020-01-15 by arkypita