arkypita/LaserGRBL #717: Optimize travel

Zoli0123, 2019-08-20:
Hello,

Could somebody help me in optimizing G0 direction?
If i would like to engraving thicker lines it takes long time because jogging. How can I reduce the jogging or optimize the process?

thx :)
[image]
[image]

[image]

Dpsspower, 2019-08-20:
It seems that you do raster image.
Try vectorize with fill.

Line by line raster only can go one direction (vertical, horizontal or diagonal) and all empty space between must be traveled.
No other way.

Zoli0123, 2019-08-20:
Hello,  It's vector style.

Dpsspower, 2019-08-21:
Ah okay. This is already filled vector.
The fill is done by raster line. So the Laser has to jump along every line.
I don't see any other way.
The software should fill each object with raster infill as a single object, then jump to the next object. But this is not how this software works. 3D slicer software works such way to save time.

arkypita, 2019-08-29:
Yes, filling of vector path is done using raster line alghoritm.
I know it is not efficient, but the code to fill vector path with vector path too is not so simple to write.

I suggest to use inkscape to generate vector filled path and then pass them to LaserGRBL.

closed 2019-08-29 by arkypita