arkypita/LaserGRBL #719: Power min.

Zoli0123, 2019-08-23:
Hello,

I have setted up minimum PWM and minimum spindle speed to S10 because the diode lifecycle is very short if it is starts from 0 to higher power. There is need base current. But vainly... After G code generating the program still contains S0 values at every G0 row and therefore the diode is turning off. How can I avoide this? I have changed manually S0 to S10 in the NC file, but it is not so precise.

Many thx

tomturican, 2019-08-24:
Hello,

my laser need a min pwm duty cycle at 6% to turn on. I´ve experimented with this problem. Normaly the $31 setting at a right value should fix this. But i got no change. Sometimes, the laser was simply of. The $31 seems without any effect. Can´t say, if this a LaserGRBL or GRBL problem or i made something wrong.
I found in the grbl config.h a setting, that fixed it for me.  In my case (grbl 1.1h) at line 350 is the entry

#define SPINDLE_PWM_MIN_VALUE 15 

You can try to uncomment this and then you have to recompile grbl. My setting 15 means a min duty cycle at 5.8%. It works fine for me. In LaserGRBL i work normaly with values from 1 to 1000 and grbl scales it from 15 to 255.

If you recompile grbl dont´t forget to change the grbl settings before ;-)

Zoli0123, 2019-09-08:
Thx for your answer :)

tomturican, 2019-09-09:
Why is S0 harmfull for diodes? If you are working in Lasermode the diode will driven with a PWM signal, means the diode switches on and of 1000 x per second with a variabnle dyty cycle....

Zoli0123, 2019-09-09:
I see. I think in case of continous PWM the diode doesn't cool down totally. But when S0 comes then it cool down. I'm red this "cold" start is harmful. Therefore they always hold a minimal base current to avoiding totally turning off.

tomturican, 2019-09-09:
Lol. Where have you read this. It´s totaly new for me.  Maybe some electronic valve technicans has written this? 
Ok. Every change of the temperature will stress every electronic part. RAM´s, processors and so on affected on this. But for this devices it is also true, that the silicon chip is mounted on a way, that will effectively spread the heat into a heatsink. This is also true for LD´s. The effect of this is, that the temperature will rise slow and this will reduce the stress for the chip.
For me personally the problem is often, that the laserdriver gives out some spikes while powering on and this is likely more harmfull for laserdiodes.
I see no problem for S0. But this is only my personal opinion. It works for me for a long time without problems. And my laser needs every chance to cool down to avoid overheating.

Dpsspower, 2019-09-11:
Temperatures much below room temperature can cause problems when the current of the diode is set over the maximum rating.
So if you start your diode in winter at 0°C, it may die because the output power will be much higher at the same current as it would be at 25°C.
The excessive output power can cause COD (catastrophic optical damage).
When using the laser in very cold enviroment, I suggest warming it up to 20°C before starting it.

closed 2020-01-15 by arkypita