arkypita/LaserGRBL #722: Laser connection

firewirehn, 2019-08-28:
I have 8w laser and eleksmana xy board.I will atach pictures and if somebody can help me with instructions how to connect this to cnc board.
[image]
[image]
[image]
[image]

StuartB4, 2019-08-28:
Can you send a picture of the controller board as well.

The black and red wires may just go to the fan, can you have a closer look to see if they do.
The thick black is probably the laser power lead, but if that does not connect straight to the eleksmana board you would have to change the plug on the end to make it compatible.

firewirehn, 2019-08-29:
https://s.amsu.ng/ZXrtgsdyPTIN

firewirehn, 2019-08-29:
How to add aditional picture

Dpsspower, 2019-09-11:
I think there is no way to drive this Laser correctly, because this Laser does NOT have a modulation input.
It just has 12V power and Ground and can only run continuously. The contant power driver is inside the housing and mostly sealed with glue.
Cheap buy = double buy :-)

StuartB4, 2019-09-11:
Hi firewirehn. How much did you pay for the laser?

firewirehn, 2019-09-11:
Around 130 us...they are sending me now modified 3 pin cable for my eleks mana board.

StuartB4, 2019-09-11:
So you should be able to use it. Thats good. 
Would be a shame if you could'nt use it when you've spent that much on it.

firewirehn, 2019-09-12:
They modified cable and I hope I can use it properly.

firewirehn, 2019-09-12:
On čet, sep 12, 2019 at 7:57, spasoje kisic<kisicspasoje@yahoo.com> wrote:   They modified cable and I hope I can use it properly.

firewirehn, 2019-09-13:
On čet, sep 12, 2019 at 7:59, spasoje kisic<kisicspasoje@yahoo.com> wrote:   
Hello Stuart,did you see modified cable which they sent to me on attached pictures

 
  On čet, sep 12, 2019 at 7:57, spasoje kisic<kisicspasoje@yahoo.com> wrote:   They modified cable and I hope I can use it properly.

StuartB4, 2019-09-13:
No, can’t see any new pictures here, just the 3 that you posted before.

firewirehn, 2019-09-13:
I will upload again.I don't know how to upload on github comment

StuartB4, 2019-09-13:
Just click at the bottom of the Comment box and select your image file.

[image]

firewirehn, 2019-09-14:
[image]
[image]
[image]
[image]
[image]
This is my laser.And they send modified cable which I have to connect to my board

StuartB4, 2019-09-14:
Ah yeah, that looks better. So it should work properly now.

firewirehn, 2019-09-14:
We will see🤣I will post one pics later for 15 w laser just too tell me am I right

Zoran543, 2019-12-08:
Hi, i can not connected PC and Laser...  :( PLEASE, help!
[image]

Zoran543, 2019-12-08:
[image]
[image]

Zoran543, 2019-12-08:
[image]

StuartB4, 2019-12-08:
You may need to install a driver for the USB. Try this first. 
https://www.srishtirobotics.com/more/blog/154-arduino-uno-r3-clone-ch340-ch341-usb-driver

Are you selecting the correct Com port.
You may need to update Microsoft .Net framwork.

StuartB4, 2019-12-08:
Make sure the red laser plug on the main board is all the way in, it looks like it's not.

closed 2020-01-15 by arkypita