arkypita/LaserGRBL #725: Images are mirrored

FoRcE721, 2019-09-01:
Hi,
   I have the Uttiny CNC Laser Engraving Machine Kit, 500mw-2500mw , I am having an issue with the images I am bringing in to burn, even the ones that came with the laser to test. 
    The images burn but the are mirrored from the original , I have not messed with any settings yet I wanted to ask first before I went to changing them. Any ideal on why this would be happening?
[image]
[image]

arkypita, 2019-09-02:
LaserGRBL use cartesian axes system. X growing right, Y growing far from user.
If your engraver does not move like this, you should configure it by inverting axes in grbl settings. Grbl settings can be configured using LaserGRBL but they are stored and managed by your engraver control board.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbl-settings

FoRcE721, 2019-09-02:
Thank you for the quick response, however I am struggling to find the solution, I do apologize as I haven’t had this setup more than 24 hours yet so u am still learning, I have tried a couple and made it worse so I reverted back but still have the same issue

FoRcE721, 2019-09-02:
Got it thanks for the help again

closed 2019-09-02 by FoRcE721