arkypita/LaserGRBL #727: Lasergrbl to Plottergrbl

garthanstey, 2019-09-06:
Some time ago there was a discussion about modifying lasergrbl to enable it to control a plotter servo. The issue was about changing the laser on/off codes (M3/M4/M5) to user supplied commands (M3S0;G4P.2) . Has anything come of this matter. I am very interested in the outcome. Thanks.
Also, is there anyway for lasergrbl to remember the co-ords of a cropped image created in the program. If I want to reload it I can never get the same image size again. Frustrating

galactyus01, 2019-09-26:
> Qualche tempo fa c'è stata una discussione sulla modifica di lasergrbl per consentirgli di controllare un servo del plotter. Il problema riguardava la modifica dei codici di accensione / spegnimento del laser (M3 / M4 / M5) in comandi forniti dall'utente (M3S0; G4P.2). È venuto fuori qualcosa da questa faccenda. Sono molto interessato al risultato. Grazie.
> Inoltre, c'è comunque che lasergrbl ricordi le coordinate di un'immagine ritagliata creata nel programma. Se voglio ricaricarlo, non potrò mai più ottenere le stesse dimensioni dell'immagine. frustrante

Ciao! sarebbe cio che sto cercando di fare! il problema e dovuto che il servo si alza quando si muove asse e quindi la linea non e precisa perche non ha il tempo di salire o scender giusto? se hai trovato qualcosa scrivimi pure.. io faro' altrettanto.. devo sistemare sta cosa..ciao e grazie

garthanstey, 2019-09-26:
Hi. I have not solved the problem completely. I started by reducing the amount of movement in the servo arm to about  30degrees. This reduces the time required for transition.
I set $Max Spindle = 100. $MIn Spindle = 0. Switch off PWM.  Set max laser power to 20 - 30 when creating gcode. I also reduced the $acceleration rate of X & Y to about 100. Not perfect!
Another idea is to use Notepad++ to find and replace every M3 with M3S25;G4P.1 and M5 with M3S0;G4P.1. (Only Notepad++ can replace 1 line with 2 lines of gcode). No $acceleration change needed for this. 
My final solution requires Vectric Aspire. I have created a Plotter Tool and gcode post processor  to make it compatible with lasergrbl.  I am able to create paths for different pen colours on one image.
Try these and let me know. Good luck

galactyus01, 2019-09-28:
Hello! I haven't forgotten you! if it is useful I have created a simple program. Just upload gcode files ... set what we want to change .. and set what should write instead of that code. .. you press button .. and save the file arranged with pause or other .. you can also modify m3 and m5 with other commands at will .. download file if you want. 
https://we.tl/t-upfUJSXifj            hello!

garthanstey, 2019-09-29:
[image]
[image]
The 2 images are A4 and created using Aspire. The way Aspire moves the pen creates interesting patterns when making "fills". Your download had expired

galactyus01, 2019-09-29:
Is beautyful! aspire program?

garthanstey, 2019-09-30:
Aspire is designed for wood engraving and milling. Not easy to learn but the results are good. It's the only program I use now. I have tried Grbl Plotter but it is not easy. There are so many settings to adjust, but it has colour identification and pen change stops.

closed 2020-01-15 by arkypita

qulet, 2020-02-16:
Can I have this program for changing the G-code in the Aspire program. The resulting file should be for Grbl1.1cprezzi _ grbl-servo.

Kuleta @ wtvk.pl

qulet, 2020-02-16:
Is this program tom postprosesor for "Aspire"?

galactyus01, 2020-02-16:
Yes my friend , i send you the file.. bye

[pause_and_config_for_gcode_plotter_.zip](https://github.com/arkypita/LaserGRBL/files/4210973/pause_and_config_for_gcode_plotter_.zip)

galactyus01, 2020-02-16:
> Posso avere questo programma per cambiare il codice G nel programma Aspire. Il file risultante dovrebbe essere per Grbl1.1 "Cprezzi _ grbl-servo".
> 
> Entra in @ wtvk.pl

this is file .rar ..... i rename in .zip for upload.. bye

[pause_and_config_for_gcode_plotter_.zip](https://github.com/arkypita/LaserGRBL/files/4210978/pause_and_config_for_gcode_plotter_.zip)

garthanstey, 2020-02-17:
I have a post processor for aspire 9 to lasergrbl. I created a v-cut tool specially for a pen plotter. Small modifications to lasergrbl configuration and it works ok.
[image]

garthanstey, 2020-02-17:
I have a post-processor for aspire9 to lasergrbl 1.1.  It needs a tool for aspire and minor modifications to lasergrbl config. 
[image]

qulet, 2020-02-18:
Dziękuję za przesłanie pliku!!

qulet, 2020-02-18:
garthanstey. Czy możesz udostępnić ten postprocesor?

garthanstey, 2020-02-19:
What language is  that?

StuartB4, 2020-02-19:
Looks like Polish.  If you copy and paste it in to Google Translate it automatically detects the language and tells you what it is.

garthanstey, 2020-02-20:
+Below is the PP file. Ctrl A, Copy and paste into Notepad and give .PP extension
+
+================================================
+                                                
+ Grbl - Vectric machine output configuration file   
+                                                
+================================================
+                                                
+ History                                        
+                                                  
+ Who      When       What                         
+ ======== ========== ===========================
+ Garth    23/07/2019 Amended for Lasergrbl-Servo Gcode Plotter
+
+ JTechP   03/28/2016 Added laser commands to
+                     every line.  Changed laser
+                     OFF command to both M03 S0 and
+                     M05.  
+ PaulW    01/20/2016 turn Spindle off during rapids
+ EdwardP  11/02/2015 Written from Grbl_mm.pp but
+                     set G20
+ EdwardP  11/02/2015 Commented out arcs as these
+                     slow GRBL performance, appear 
+                     interpolated anyway
+================================================

+------------------------------------------------
+    General Notes               
+------------------------------------------------

+  Designed for Spindle speed to control a Servo on Pin D11.
+  
+  Here the Pen up/down are M3S0=Up, M3S30=Down, followed by G4P0.1
+  for the servo delay. For versions 1 and above,
+  set GRBL configuration to $30=100,$31=0. Set Vcarve spindle
+  speed to 100rpm, VTool Flat width to pen tip Diameter (.2mm?)
+  Tool diameter 1mm, Angle 5deg, Pass = material depth -.1mm
+  for 1mm pass. Stepover to 95% for fills. Feed = Pen speed (2000mm/m?).
+  Most servos jump to S0 at power-on and S0=Up stops unwanted
+  pen marks on the target. The Servo can be reversed or range
+  altered by changing the current "30" value which produces 
+  about 50-60 degrees movement. Both the Header, Footer,
+  plunge_move and plunge_retract and G4 delays can be
+  amended as necessary.
+================================================

POST_NAME = "GRBL PLOTTER (MM) (*.gcode)"
 
FILE_EXTENSION = "gcode"
 
UNITS = "MM"

SPINDLE_SPEED_RANGE = 0 30 0 100
+    Sets Spindle stop, Spindle Max, Min RPM, Max RPM:
+    [Actual RPM usually in thousands so "100" should always
+    create S0 and Max S30]
+    Create dedicated "Plotter" VBit as detailed in the notes
+    and use in a V-Carve Toolpath for output
+    The "offset" option produces nice "fill" effects.
+------------------------------------------------
+    Line terminating characters               
+------------------------------------------------
 
LINE_ENDING = "[13][10]"
 
+------------------------------------------------
+    Block numbering                             
+------------------------------------------------
 
LINE_NUMBER_START     = 1
[... 123 lines cut ...]
+  Commands output for the plunge move
+---------------------------------------------------

begin PLUNGE_MOVE

"M3S30"
+ Pen Down
"G4P.1"
+ Pause for Pen Up/Down

+---------------------------------------------------
+  Commands output at the end of the file
+---------------------------------------------------
 
begin FOOTER
 
"M3S0"
+ Pen Up
"G4P.1"
+ Pause for Pen Up/Down
"M5"
+ Spindle stop
"G0[XH][YH]"
+ Home
](url)

qulet, 2020-02-20:
My plotter works on this software from this link:
https://github.com/bdring/Grbl_Pen_Servo
Very good build, you don't need to change anything in the G-code.
https://drive.google.com/file/d/1CV2XThJwnwFo9rxF4doIsp9TD-Az5ZrZ/view?usp=sharing

qulet, 2020-02-20:
garthanstey
How to get layers in Vectric Aspire. when importing a layered SVG file. I'm not asking about manually selecting vectors and moving to a new layer. For example, LIGHTBURN does such a neat import of layers?

garthanstey, 2020-02-21:
I'm sorry but I don't use layers. I ask Aspire to break the image into colours and create a different toolpath for each colour. That was how I created the Peacock.

qulet, 2020-02-21:
garthansey

Many thanks for the hint !!!

qulet, 2021-05-21:
How to generate G-code path with vector fill for pen plotter in Aspire? What machining strategy is needed?

qulet, 2021-05-21:
garthanstey
  Can you provide this post processor for peacock fill drawing? What strategies are used to generate G-code padding? Your g-code commands for servo UP and servo Down.

qulet, 2021-05-22:
Question not valid. I have read the GRBL PLOTTER (MM) postprocessor comments and I have followed the comments contained therein. Vector filling started to work. Thank you !!!!!

garth1664, 2021-05-30:
Quiet - I have a Nano programmed with a Servo Hex file which came from Github. This plotter uses the spindle speed to operate the servo. I set the spindle to 100rpm $30=100:$31=0:$32=0. This covers 180Deg on my servo. The servo needs to function on the PP (Plunge_Move) with GCode "M3S40" followed by "G4P0.3". The PP (Retract_Move) needs Gcode "M3S0" followed by "G4P0.3". These last 2 also need to be in the PP Header and Footer for pen Safety.
The Aspire program needs a dedicated End Mill tool. I have several, 1 for each nib thickness. set the diameter to the nib, set plunge depth to 0.2, stepover depends on the ink bleed of the pen/material, spindle = 100rpm, Feed varies with ink travel to nib, try 2000mm/m. Plunge rate not too relevant but 1000mm/m works.
Choose a Pocket path for fills and all should be well. Let me know.

qulet, 2021-05-30:
[image]
My first fill vector generated in Aspire Thank you for your help! ....

garthanstey, 2021-05-30:
Nice work! Good effect.
If I might comment. You could try a 90% stepover to reduce gaps between lines, or if ink flow is causing it, try reducing the feed rate so that the ink in the pen reservoir can reach the nib faster than it's being used. If you get blobs (bleed) of ink where the pen lands, reduce the delay in PP G4P to 0.2 or less if your pen has a very short drop.
The peacock was a line drawing from the internet. Using a paint program I filled segments with colours. They don't have to be the final colour. The pen you use decides that. But it's easier to see the final possibilities. 
Aspire converts the bitmap to vectors. Reduce the number of colours to do just the colours you need. Select the colour you want to plot and create a tool path. Name it, put the colour in the name and save it. Repeat until all colours are done. Don't forget the original black outline or maybe make it a different colour.
Once again, great start. Where to now?

qulet, 2021-05-31:
Now I will try to generate a c-code with colors according to your recipe. I will show off the end result.
Greetings!

qulet, 2021-05-31:
[image]
How can I get a color gradient?
Tweete_go's cheeks

garthanstey, 2021-05-31:
Is the bird your own creation?
I don't think a colour gradient is possible. Ink Flow is not controlable and mixing colours before they dry is not practical.

parthshah01234, 2023-01-27:
garthanstey
Can you share postprocessing?

imen-sys, 2024-11-14:
# [✅▶▶ DOWNLOAD Vectric Aspire PRO](https://shorturl.at/YEPVq)

# [✅▶▶ MIRROR Vectric Aspire CRACK](https://shorturl.at/YEPVq)

.

.

.

# [✅▶▶ DOWNLOAD Vectric Aspire PRO](https://shorturl.at/YEPVq)

# [✅▶▶ MIRROR Vectric Aspire CRACK](https://shorturl.at/YEPVq)

.

.

.
///